opentrack/opentrack #500: Czech location

petsild, 2016-11-23:
Hi,
I'd like to do Czech location for Opentrack. I have to use some special sw? Pet

MathijsG, 2016-11-27:
I just forked this repo and grabbed the Russian file from here: https://github.com/opentrack/opentrack/tree/unstable/gui/lang and editted it with a text editor (like Notepad++) and then did a pull request.

sthalik, 2016-11-29:
See https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa for translation stubs.

petsild, 2016-12-26:
Hi,
i have translated the Gui into the Czech language. Probably will want to, where appropriate correction of the text. Like it before the release of the big test. Write please what can I do thank you. Pet
[cz_CZ.txt](https://github.com/opentrack/opentrack/files/673211/cz_CZ.txt)

sthalik, 2017-02-27:
@petsild @MathijsG you now have .ts files in all releases in the `source-code/` subdirectory.

sthalik, 2017-02-27:
@petsild please send as pull request.

closed 2017-02-27 by sthalik