opentrack/opentrack #501: Opentrack freeze on most actions

mysticfm, 2016-11-26:
For me, Opentrack (both older builds of 2.3 and the latest one) invariably goes into a "Not Responding" status for at least a minute, and occasionally hangs permanently, when I take any of the following actions (and perhaps others, but these are the ones I have run into):

1) When starting the Opentrack application
2) When starting tracking (with the Start button in Opentrack)
3) When binding a keystroke in the Options dialog

Note that stopping tracking and shutting down the application are instantaneous.

I'm running on Windows 10 Home, version 10.0.14393.  My head tracking device is an EDTracker Pro. The only antivirus on my system is Windows Defender.

This has been a longstanding, frustrating impediment to my making much use of Opentrack.  If there is anything I can do to help resolve this at last, please let me know.

Len62, 2016-11-26:
Are you running OT in Admin mode? EDTrackerPro is 3 point correct? Make sure to disable the translation axis' in OT.

sthalik, 2016-11-27:
@Len62 this isn't relevant, sorry.

@mysticfm Please install "Visual Studio 2015 Community" and report back. I'll give you files and instructions on how to extract relevant debug info.

Len62, 2016-11-27:
@sthalik I thought as much. BTW P15 is working fine w/my set-up.

sthalik, 2016-11-27:
@mysticfm to get few things out of the way

- are you using multiple displays? If so, we might have another bug on multiple display configurations.
- do you have an Android phone plugged in? If so, try unplug/replug it. I have this problem with regard to the phone sleeping and timing out various operations.

mysticfm, 2016-11-27:
I am not using multiple displays. I have a single GTX 980 Ti driving an Acer 2560x1440 display with G-sync.

I don't have an Android phone (or any phone) plugged in.  I do have the following USB devices plugged in:

- Corsair K95 RGB mechanical keyboard
- Microsoft optical mouse
- CH Products Fighterstick Pro
- CH Products Pro Throttle
- CH Products Pro Pedals
- EDTracker Pro (of course)

I have Visual Studio 2015 Community Edition downloading and installing as I type this ... should be finished installing it within another hour or so.

mysticfm, 2016-11-27:
As an additional test, I just tried launching OT after unplugging all of the CH Products controllers that I'd listed in the previous message.  It made no difference ... the delay in OT's initial startup as well as the start tracking operation was exactly the same (a little over a minute in each case).   Obviously it isn't going to be practical for me to attempt starting and using OT with the keyboard or mouse unplugged, so I haven't endeavored to test that.

I will also note that Task Manager reports next to no CPU usage by OT while it is in this "Not Responding" state.  So I'd assume it is indeed blocking on some external request, rather than stuck in some sort of semi-endless loop.

I'd like to elaborate that the minute of "Not Responding" status during the initial OT startup occurs before the UI has been displayed at all ... there isn't even an icon on the taskbar for OT yet for that first minute, although I can see the OT process in Task Manager.

Finally, VS2015 is installed on my system now, so I await your further instructions on debugging the issue.  Thanks!

sthalik, 2016-12-24:
Drop me a stack trace of all running threads at the time it's frozen.

sthalik, 2017-02-27:
@mysticfm do you have multiple displays? There was a bug with multiple displays freezing the software but it works now.

mysticfm, 2017-02-27:
I don’t have multiple displays.

Also, I haven’t done anything with the stack trace you requested because I don’t feel I have enough information to proceed upon.  Won't I at a minimum need a local debug build of opentrack in order to do that?  You didn’t mention anything about that in your previous emails.

mysticfm, 2018-02-08:
No solution, just living with the problem. Rather than trying to work with me to identify the issue, the developer basically  handed me a link to the debug build and told me to work it out for myself, and I'm afraid I have more important things to do.

sthalik, 2018-02-09:
Are you using "GIGABYTE OC GURU"?

sthalik, 2018-02-09:
@mysticfm I can't do anything with a backtrace. This problem is specific to your system, or very few systems. There's no chance of me reproducing it.

Neuse87, 2018-02-09:
Nope never used OC Guru

sthalik, 2018-02-09:
- Download "sysinternals process explorer".
- Scroll to your `opentrack.exe` process. Don't double-click on it, just select it.
- Wait for it to freeze
- Enable "view -> lower pane" in procexp
- In "view -> lower pane view" select "DLLs"
- The list isn't sadly exportable so make multiple screenshots with all of the entries. **Sort by full path, not name itself**.

Neuse87, 2018-02-13:
Du you have an Oculus rift? @ mysticfm

Neuse87, 2018-02-14:
Just asking because removing all the oculus devices from my pc somehow improved the issue.

closed 2018-08-04 by sthalik