opentrack/opentrack #502: How to use Aruco

izams, 2016-11-27:
Hello there,

I am new to this. Just installed the latest version and now have few issues that needed your expert advise.

1- I have read the wiki on how to set-up using Aruco as tracker. As I started the application & selected 
    Aruco as tracker, why there were no movement or it did not response at all, as I waved the paper 
    (printed from the file attached) in front of my camera. 
    What can possibly be the problem?
    
2- Can I just use the existing webcam as tracker? If so how can I set it up? 

FYI, before I submitted this issue I had thoroughly search evrywhere in the net for an explanation or tutorial but found a few with very little information which I was unable to understand. If this matter has been brought up kindly link me to the appropriate page & I am sorry for that. 

I intended to use this application for my flight simming (FSX). To the developer of this great tool, I really appreciate this piece of work very much.  

Hope you guys could assist me in resolving this issue. Thanking you all in advance. 

Cheers,
Izam

Emton, 2016-11-27:
Did you ever see the shape get picked up (shown with a purple outline of the shape) How did you print the shape is it on fairly rigid board? is it highly glossy (glossy will be bad) Try again at different angles until you see if the shape gets picked up at all.

sthalik, 2016-11-27:
You need some spacing around the marker. Leave some white paper around the symbology.

izams, 2016-11-28:
TQVM for the early response. I'll try to answer as good as I can as honestly I am still trying to familiarize with this. 

Emton,
- Sorry but I did not get you when you mentioned about that 'red outline' thingy. 
- I printed our from an A4 size paper & have not paste on cardboard yet. Just tested from that paper itself
- I can confirmed that there were nothing blocking the shape & it was not glossy. 
- I have tested all angle possible with same result. I could see the video feed, the cam worked & am using a webcam with LED at its side.

sthalik,
I printed out from the folder under the name test3.png. It was on an A4 paper & left some white area around the paper. 

Hope the information above would suffice. Thanks again for your kind attention & action. 

Regards,
Izam

sthalik, 2016-11-28:
@izams send a single camera frame with the marker visible, please.

izams, 2016-11-28:
sthalik,

Did you mean a screenshot of my webcam feed with the paper up OR only the picture of the marker?

I can do that but will only be able to do it tomorrow evening when I am back home. Hope you understand. 

Thanks for your time. 

Cheers,
Izam

sthalik, 2016-12-28:
@izams webcam feed. A full-resolution frame, not inside the opentrack window.

izams, 2017-01-13:
Hi sthalik,

Sorry for the late response. At the moment I managed to use opentrack via smartphone/android. The only thing now is to figure out how to calibrate.

In the meantime I leave the Aruco option & try out the UDP/smartphone opition. 

Thanks again for your kind response.

Cheers,
Izam

closed 2017-12-02 by sthalik