opentrack/opentrack #503: Hotkeys for Switching Output Mode

kkotowick, 2016-12-01:
With support for Oculus runtime 1.4.0, I've been using this software every day. Thanks!

I'm trying to use it in a game where I need to quickly change the output mode from "Freetrack 2.0" to "Mouse Emulation". I can change this in the dropdown in Opentrack, but I need a way to be able to quickly do it from in the game, without changing window focus. Would there be any way to add a hotkey binding that could toggle the output mode?

sthalik, 2016-12-01:
No, there's no way to change the modules used while tracking is active.

See #492 but it's gonna take some time before it's implemented.

closed 2016-12-01 by sthalik

vinKaiZen, 2016-12-01:
You can do it when using Opentrack & FreePie (http://andersmalmgren.github.io/FreePIE/) with my phython script
   
```
 #Use T to toggle head tracking  on/off 
def update():
    yaw = trackIR.yaw
    pitch = trackIR.pitch
   
    deltaYaw = filters.delta(yaw)
    deltaPitch = filters.delta(pitch)   
     
    if (enabled):
        mouse.deltaX = deltaYaw*multiply
        mouse.deltaY = -deltaPitch*multiply
		
if starting:
    enabled = False
    multiply = 65
    trackIR.update += update

toggle = keyboard.getPressed(Key.T)

if toggle:
    enabled = not enabled   
```

Just insert and start script with F5.
With the "T" Button you can switch on/off mouseview when using Headtracking with Opentrack
If you wanna change Sensivity just edit the multiply value "65". (higher is more sensivity)

sthalik, 2016-12-18:
@vinKaiZen it's trickier than that. For each movement, there are two paths for the mouse to take and it must choose the shorter one.

vinKaiZen, 2016-12-19:
Hmm, it works absolutely fine 4 me since last 14 month. its realitive to your acutal mouse position.