opentrack/opentrack #504: Cap model sometimes loses calibration point and inverts

peterek1983, 2016-12-01:
Why when I invert my cap out of camera view for a moment then program lost calibration and everything is mess all axes ?

How to fix that ?

sthalik, 2016-12-01:
It's the "dynamic pose resolution" option. A fix for the situation is to cover a single LED from the camera for its timeout's duration.

A solution for this would be code able to differentiate the points without having known where they were previously. I have thought about this but I'm not entirely sure it's possible. I'll give it a try in the indeterminate future but don't wait on it.

peterek1983, 2016-12-01:
Cover one single LED it's not a solution for me :) 
Why in Freetrack there is no problem like this, this should be first on list to fix, with this bug opentrack it's no useable for me, sorry :(

In Polish - Proszę zajmij sie tym ;) byłoby super jakby to się udało poprawić bo to jest nie do przyjęcia żeby takie cuda sie działy jak sie odejdzie od monitora i wróci. Pozdrawiam

sthalik, 2016-12-01:
Nie, zakryć aż się zresetuje i puścić.

Po pierwsze primo, ja nie mam z tego kasy. Po drugie primo, nie zawsze jestem w stanie pisać. Po trzecie primo, kto inny może napisać i wysłać pull request.

sthalik, 2016-12-01:
Po czwarte primo, to nie jest bug tylko nieprzyjemne ograniczenie.

sthalik, 2016-12-01:
No i jeszcze dodam, że to nie jest trywialne rozróżnić punkty - jak spojrzysz w górę czapką to środkowy jest na górze a nie na dole. Jak spojrzysz w bok to środkowy nie jest na środku, etc.

sthalik, 2016-12-01:
freetrack miał też takie samo rozróżnianie punktów. Miał też szukanie środkowego na podstawie jego wysokości w ramce od kamery ale to nie zawsze jest poprawne. Ciężko się czyta tego Pascala ale tak jest.

Ustawisz sobie stabilność ruchu parametrami Acceli. Daj "rot nonlinearity" na 1.1, sensitivity na mniej, deadzone na 0.03, smoothing na 2-5 ms. Nawet przy bardzo zaszumionych danych wejściowych zadziała jak dobrze pokręcisz gałkami.

Oczywiście najważniejsze jest żeby model miał wymiary zgodne z klipsem.

peterek1983, 2016-12-01:
Super wielkie dzięki za sugestie odnośnie ustawień, czapkę mam zrobioną z odblaskami myślę wzorowo, czyli odległości domyślnie jak dla Capa 3 punktowego w programach + A4Tech PK333E. Testuje obecnie w American Truck Simulator :)

peterek1983, 2016-12-01:
Kombinuje na rózne sposoby i nie moge uzyskać płynności Freetracka, tzn ogólnie nie jest zle, ale Opentrack ma pewną tendencje do przeskakiwania minimalnego w ruchu niezależnie co by sie nie ustawiło w ustawieniach Acceli.

Tutaj mozesz zobaczyć o co mi chodzi ustawiony minimalny łagodny deadzone na początku na dole osi i kulka w programie na krzywej chodzi po tych idealnie gładko jednak w grze jest mały skok na przejściu jakby schodek
https://youtu.be/g4Kyk18VnFE

We Freetracku podobna krzywa i nie ma takiego przeskoku jest tylko jakby w tym miejscu łagodne zwolnienie 

Nawet jak ustawie liniowość całkowitą na krzywej to podczas ruchu dostrzegam mikroskopijne przeskoki jakby przyspieszanie i zwalnianie na osi, oczywiście mozliwe ze mój ruch głową wtedy nie jest idealnie równy ale nie moge tego skorygować ustawieniami Acceli aby było ultra łagodnie i płynnie jak we Freetracku.

Nie orientujesz sie co brakuje Freetrackowi ze nie chce w niektórych grach działać, od taki Dirt 3, w OpenTracku działa we Freetracku juz nie, pomimo ze mam dograny plik NPClient64.dll.

sthalik, 2016-12-01:
To może być co innego. Prześlę builda niedługo.

peterek1983, 2016-12-01:
OK super dzieki! czekam na builda...

sthalik, 2016-12-02:
Builda z dzisiaj - https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa?dl=0

W facetracknoir do dzisiaj nie naprawili, że protokół, filtr etc. działają jedynie 66 razy na sekundę.

peterek1983, 2016-12-02:
Dzieki piekne, zaraz biore sie za testowanie

peterek1983, 2016-12-02:
Po szybkim porównaniu obu wersji powiem szczerze ze nie dostrzegam jakis różnic w ruchu, niestety nadal patrząc np. w kokpicie samolotu na zoomie na instrumenty rozgladając sie po ukosie prowadzenie nie jest równe po ukosie odczuwam takie jakby schodki w ruchu jakby coś przyciągało osie w ruchu ukośnym raz albo dla osi Yaw raz dla Pitch, mam nadzieje ze wiesz o co mi chodzi. Sprawdzam Freetracka i tam jest łagodniej nie ma czegoś takiego.
Bede jeszcze testował.

sthalik, 2016-12-02:
Daj screena z ustawieniami acceli.

peterek1983, 2016-12-02:
[https://s13.postimg.org/brqt65klz/Bez_tytu_u.png](url)

sthalik, 2016-12-02:
Nonlinearity na 1.15 do 1.2. Deadzone na 0.03 do 0.06.

peterek1983, 2016-12-02:
OK :) tak tez ustawie zaraz i potestuje

closed 2018-05-14 by sthalik