opentrack/opentrack #506: How to save data to RAM

PintsizedSix40, 2016-12-03:
Hi, I'm wondering if it is possible to save the Yaw Pitch and Roll values to the RAM so I could access them using another program.
Thanks.

sthalik, 2016-12-04:
Use the freetrack protocol as it is right now.

PintsizedSix40, 2016-12-04:
But than how would I get the values?

sthalik, 2016-12-04:
See https://github.com/opentrack/opentrack/tree/unstable/freetrackclient

PintsizedSix40, 2016-12-04:
sthalik:
> See https://github.com/opentrack/opentrack/tree/unstable/freetrackclient

What do you mean?
I'm trying to get RAM addresses for the tracking values.

sthalik, 2016-12-04:
See the linked directory for mapping name and data layout.

PintsizedSix40, 2016-12-05:
Where?

sthalik, 2016-12-05:
The link I gave you.

PintsizedSix40, 2016-12-06:
> sthalik:
 See the linked directory for mapping name and data layout.
 

> PintsizedSix40:
Where?

I mean where in https://github.com/opentrack/opentrack/tree/unstable/freetrackclient where do I find the mapping name and data layout?

sthalik, 2016-12-06:
In the source file and the header file, respectively.

closed 2016-12-24 by sthalik

PintsizedSix40, 2019-03-17:
Lol sorry for being this stupid.