opentrack/opentrack #507: UDP sender działa tylko od -180 do 0, brak 0-180

topiq77, 2016-12-07:
Witam. Jak w temacie. UDP sender ma zakres obrotu na osi YAW tylko od -180 do 0, następnie przeskakuje do -180. Sprawdziłem poprzednie wersje, ostatnia działająca to RC99P14, od wersji P15 błąd już występuje, ale nie miałem niestety okazji wcześniej tego wyłapać (brak czasu itp.). Problem identyczny jak kiedyś w P12, który już wstawiałem na filmie z porównaniem do P7 https://www.youtube.com/watch?v=olLn5wWB_og
Pozdrawiam.

sthalik, 2016-12-08:
Wartości na wejściu yaw nie są poprawne. Powinny być między -180 a 180. To błąd w firmware.

sthalik, 2016-12-08:
Nie wiem czy uznać to za błąd skoro to firmware jest dość popularne.

topiq77, 2016-12-08:
Ok żeby nie było nieporozumień, bo nie wiem co ma do tego firmware telefonu.
Filmik z działania UDP Sender na finalnej wersji OpenTrack 2.3.0 https://youtu.be/tFglmE1Irc8 , jak widać RAW data na osi YAW jest wszystko w porządku. Żyroskop w telefonie po skalibrowaniu bez problemu robi obrót o 360 st. Program przetwarza tylko dane od 0 do -180.
Tu mamy prawidłowo działający UDP Sender https://youtu.be/CAEt3kxkVdU wersja 2.3.0 RC99 P14 ostatnia prawidłowo przetwarzająca dane z osi YAW w zakresie od +180 -0 -180 st. Ten sam firmware telefonu, ten sam żyroskop, skalibrowany 5 minut wcześniej dla wersji finalnej. Jak dla mnie to jest bug UDP Sendera od wersji RC99 P15, bo tam też tak samo źle działa.
Pozdrawiam. Mam nadzieję, że dostatecznie to jasno wytłumaczyłem :)

sthalik, 2016-12-08:
opentrack przyjmuje konwencję kątów -180 do 180. Tamten tracker wysyła 0 do 360.

Mogę dodać "troubleshooting" do opcji trackera.

topiq77, 2016-12-09:
Nokia Lumia 820 WP 8.1 Update 2 (8.10.15116.125) Układowe 3051.50009.1451.1013.
Soft na tą Lumkę OpenTrack Head Tracking 1.2.0.4.
Tylko czemu na RC99P14 działa poprawnie? Póki co używam tej wersji.

sthalik, 2016-12-10:
Napisałem do kolesia.

MopheusDG, 2016-12-18:
I don't understand Polish but I have some problems with this version of OTrack and UDP Sender too, not exactly the same, in my case when I turn right, the YAW angle works perfect, when I turn left, the YAW angle goes well till 10-20% then just jump to -180. Any ideas ? Is this related ?

sthalik, 2016-12-18:
What's the raw data range the tracker receives? 0->360 as well?

sthalik, 2016-12-18:
@MorpheusDG Please confirm the range you have so I can fix this and release 2.3.1 immediately after.

sthalik, 2016-12-18:
@MopheusDG Please confirm the range you have so I can fix this and release 2.3.1 immediately after.

MopheusDG, 2016-12-18:
It's strange, I've been trying to reproduce my problem on my laptop and cannot. It's only happening on my game PC and cannot test right now. Let me clear some things out:

1) My phone doesn't have Gyro (Old Samsung Fame phone). On the Freepie app information from debug of course show Gyro on 0. Anyway, I can rotate 360 degrees without no problem and seems accurate on my laptop. Now on the Game PC when I try to rotate, I only can rotate using my phone in horizontal pointing to my left or right, because if I use it pointing my screen, when I rotate suddenly I get this problems. But debug data sees to be fine, no spikes from what I see.

2) I was thinking to got all values to default on my PC and try again before reporting later, is there any to reset all OTrack settings and be sure to start over ? cause I've been playing with to many betas, etc... and using the portable version if I delete the folder, still have my old settings after,

3) Is it normal that head movement using a phone with NO gyro on games (Live for Speed or Euro Truck 2) look a little slow or unresponsive ? Is it better with Gyro or the delay or slowness is normal using udp ?

Thanks, will try to get more info later. Sorry.

MopheusDG, 2016-12-18:
Ok, new findings. I was able to reproduce my problem. When ORIENTATION on the debug mode on Freepie goes from higher than 3 it goes right to -3 (seems that happen after giving a full 360 degrees rotation). I think it may be possible that if you dont have gyro and use orientation instead (compass ?) after moving from 0 to 3 it goes back to -3 and OTrack goes crazy.

Another thing I notice is that when I use Pitch, it works perfectly but my Roll data is changing too, even if I'm not rolling. This help me find another issue, even if I select DISABLE under OPTIONS for the ROLL axis, it keeps sending the information to the game, so I think those DISABLE options may not be working ?

Thanks and keep the good work, this tool is AMAZING !

sthalik, 2016-12-18:
@MopheusDG what's the raw data? Also not having a gyro is a show-stopper.

MopheusDG, 2016-12-18:
Raw data for each axis:

Yaw Raw: mine is center in 90 (my phone is rotated to avoid issues with bad YAW). I can go to the left, negative till -180 and I can go right till 180.  So basically 90 y my center in RAW and I can rotate both sides, left if I go to far (more than -180 it switchs to 180 and goes a little crazy). I probably can fix this assigning +90 to the axis under options).

Roll Raw (problematic): Assigned to PITCH in my case (cause my phone is rotated) 0 is center. Up and down goes fine, Raw moves fine between -90 to 90 and PITCH RAW stays in 0. BUTTTT, even if RAW data for PITCH (Assigned to GAME DATA ROLL) is in 0, GAME DATA ROLL starts changing to 10 or -10. Example:

RAW Pitch: 0                  Game Data Pitch: 45
Roll Pitch: 45                  Roll Data: -8 ???

And by the way, this happens EVEN when RAW source is DISABLED. Also noted this happen even if axis are not crossed and the difference is more evident when going negative values, positive seems more ok.

MopheusDG, 2016-12-18:
The only way I found to fix this is increasing the ROLL axis deadzone so that it only moves when values are higher than 10% on raw data. But again, it keeps reporting data to game even if the axis is disabled. Hope this helps.

sthalik, 2016-12-19:
I'll add an option to select a constant value that's added to the raw data: 45, -45, 90, -90. That should work for you.

As for roll input, zero it out in the mapping window. Centering causes that.

sthalik, 2016-12-24:
You already have the option to add a raw value to yaw. Under "add to axis" in UDP tracker settings...

MopheusDG, 2016-12-24:
Thank u VERY much ! Will test later but thanks, best Christmasts present ;)

sthalik, 2016-12-24:
Have a good one!

topiq77, 2016-12-26:
OK. Pogrzebałem w ustawieniach UDP Sender, dodałem w Add to axis, Yaw: -180, zmodyfikowałem lekko Mapping i daje radę ;) Dzięki za podpowiedź. Wesołych Świąt i dalszego rozwoju wspaniałej aplikacji jaką jest OpenTrack.

sthalik, 2016-12-26:
Dziękuję. Również wesołych świąt :santa: :cat2:

closed 2016-12-26 by sthalik