opentrack/opentrack #509: aruco printing

denisdgguga, 2016-12-09:
i want to stick aruco in the middle of my [vrbox](http://happyvr.eu/review/review-vr-box-ii/) but i want to know

1. how big need aruco to be
2. can i print it in a A4 paper
3. do i print it like any other photo
4. where to find aruco
5. any other tips guys

Emton, 2016-12-10:
1. It should be probably at least 3x3in - 4x4in with some white border around the edge.
2. You can print it on anything that is white and non-glossy but it can't be flimsy so if it's paper you have to mount it to something rigid and flat, however you do that it should be neat (no glue bleeding through and crinkling the paper) If you have to "glue" it use spray adhesive you can get at craft store it's made for attaching paper to board. I skipped that and printed mine on the cardboard box the PS3eye cam came in but make sure the printer can handle the thickness of the board. 
3. Don't print it like a photo unless you want to experiment using matte photo paper, don't use glossy. Print using the darkest ink settings your printer can handle (don't use low quality printing)
4. Look through the opentrack folder for the .png image file
5. https://github.com/opentrack/opentrack/wiki/Aruco-tracker

sthalik, 2016-12-10:
You need to also preserve some of the white border around the marker. Don't cut it too close to the black marker. The border is an integral part of it.

closed 2017-12-02 by sthalik