opentrack/opentrack #510: Point Tracker with camera above

cutzmf, 2016-12-12:
To be clear:
I have DIY HMD and camera mounted above me - I want 360 degrees yaw. I remapped axis Roll -> Yaw

Now I have an issue with new yaw axis(orig Roll) - it has error at (and above) 90 degrees.
Is it possible to make a camera mount switch position (front/upper)

sthalik, 2016-12-13:
Do you have "dynamic pose resolution" set?

Does the model center, the red cross, jump abruptly at roll +- 90 deg?

cutzmf, 2016-12-13:
1. "dynamic pose resolution" - i'll update later (but seems to be off)
2. model center really jumps abruptly at roll +- 90 deg

sthalik, 2016-12-14:
Set dynamic pose to on with a sensible timeout -- no less than 500 ms.

cutzmf, 2016-12-15:
I've switched on dynamic pose resolution.
With 500ms timer - i've got errors above 120 degrees.
Tried 700ms - got better angle (up to 150).
It seems to me I need to set  timer at 1000ms to get rid of errors at 180 degrees motion =)

Thank you!
I appreciate your work! =)

sthalik, 2016-12-16:
I don't think this the correct solution but glad it at least works for you now.

If dynamic pose confuses itself, remember to cover one LED for the dynamic pose timeout's duration. It'll reset itself after you uncover it.

closed 2016-12-16 by sthalik

cutzmf, 2016-12-16:
By the way, would it be too hard to implement feature "vertical mounted camera"?

sthalik, 2016-12-16:
What for?