opentrack/opentrack #511: Suggestion : allow multiple joystick for different axis to emulate 6dof

albertolg, 2016-12-15:
@sthalik I use opentrack with edtracker, but as you know edtracker it is limited to 3dof, yaw-pitch-roll. I would like that opentrack could map another jpystick to be used to the other axis, example, edtracker input for yaw-pitch-roll, the rudder on my hotas for z or y so i can move forward and back or up and down. I know this exist in freetracknoir but I like much more opentrack and would like use it here. Thanks

sthalik, 2016-12-15:
It's a good suggestion.

I'll try to implement multiple tracker output able to run the same tracker multiple times. But if that's too complex, I'll fall back to your solution.

Let's keep this issue open regardless.

albertolg, 2016-12-15:
great, thank you :-)

albertolg, 2016-12-26:
I want add something... would be nice to be possible to assign also a non axis command, like a keyboard button or the button of any joystick

sthalik, 2016-12-26:
What do you mean by that?

albertolg, 2016-12-26:
for example... I'm still referring to the Joystick input. Right now you can select the single joystick and map his axis to x-y-z-yaw-pitch-roll, ok? If for any axis you can choose a different input?
x=axe X of joystick1
y=axe Y of joystick1 or joystick2 button1 or keyboard A (this will also need to implement a sort of virtual axe, that can vary pressing the assigned button)
z=axe X of joystick2 or keyboard Z or anything like previous
... and so on

maybe this is too complicated to implement right now?

sthalik, 2016-12-26:
We'll have python support eventually. This is too complex to implement in the base UI. Python will be able to create own UI.

sthalik, 2017-02-21:
See #492

closed 2017-02-21 by sthalik