opentrack/opentrack #512: Oculus Rift runtime 1.10.1 HMD

huliqan, 2016-12-16:
can You add the last 1.10.1 runtime oculus rift to opentrack 2.3?
**Oculus Rift runtime 1.10.1 HMD** needed

sthalik, 2016-12-16:
Sure just give it some time.

sthalik, 2016-12-16:
I'm gonna make a single Rift tracker with selectable SDK.

huliqan, 2016-12-16:
it"s very cool!!!

huliqan, 2016-12-19:
oops...
in opentrack 2.3 > input > Oculus Rift runtime 1.4.0 - HMD 
all works to last runtime 1.10.1!!!
everything is working !!! just my antivirus blocked the process :)))

sthalik, 2016-12-19:
@huliqan stated that it works now using the 1.10 SDK. No need to make any changes.

"Persistence" is for older models with gyro drift.

closed 2016-12-19 by sthalik

abhinaynatti, 2018-05-17:
it says" error occured while loading tracker Oculus Rift runtime 1.4.0 -- HMD" when I put it as input.Can you tell me what I missed here .Thank you.

sthalik, 2018-05-17:
What's your API version?

abhinaynatti, 2018-05-17:
2.3.10

sthalik, 2018-05-17:
No, Rift SDK version. It was said that it works with 1.10 SDK, is there a new version?

abhinaynatti, 2018-05-17:
Actually what I need is I need to analyze the  data of head movements(pitch etc. ). The values are getting stored in the csv file.  I don't actually know how it is tracking .

abhinaynatti, 2018-05-17:
So I don't know how it takes into consideration of the oculus.

sthalik, 2018-05-17:
It's `libopentrack-tracker-rift-140.dll` or another API version you chose. See the `tracker-rift-140/` subdirectory for the source code.

abhinaynatti, 2018-05-17:
Hey can I ask from the start?This is more like a doubt than an issue. What I basically want is to analyze the  data of head movements when I'm using my VR headset . So what input should I be using and what software should I download. Sorry for the trouble.

sthalik, 2018-05-17:
Read the "raw" column. You can draw a plot with libreoffice.