opentrack/opentrack #513: ArmA 3 64Bit Issue Question

Suppresed, 2016-12-16:
Arma 3's Dev Branch had an update that made the game 64 bit. Headtracking is said to not work, can anybody confirm this?

sthalik, 2016-12-16:
We support headtracking for 64-bit games.

See if it works without BattlEye. If it does, please send our NPClient64.dll to BattlEye.

Emton, 2016-12-17:
I tried it with Battleye disabled but it still didn't work.

sthalik, 2016-12-18:
Does it work for anyone at all, with trackir or something else?

sthalik, 2016-12-18:
You can tell bohemia to look for "NPClient64" and use the same head tracking API as they did before.

petsild, 2016-12-26:
ArmA 3 64Bit with OT work without problem.

sthalik, 2016-12-26:
@petsild does it work with BE?

petsild, 2016-12-26:
Tested on DB 1.67.139743

okaryna, 2017-03-18:
I confirm after x64 patch of Arma3 opentrack stops work in version 3.4.1.
I am not sure it is Arma problem or opentrack settings.
Solution: in "FreeTrack 2.0 Enchanced "options set "User TrackIR, hide FreeTrack".

sthalik, 2017-06-18:
Works for me also. BattlEye enabled.

closed 2017-06-18 by sthalik