opentrack/opentrack #514: Open track stopped working on games

Tafkaa, 2016-12-16:
I've been using opentrack (with aruca paper tracking) for a while now and up until this morning it worked fine. Now, all of a sudden, none of my games seem to recognize that opentrack is running.  I have no idea what changed or how to fix it. Already tried reinstalling but that didn't help. To clarify; the tracking itself works fine (the octopus moves as expected) but in game there is no movement at all anymore.

Any ideas how to fix this?

sthalik, 2016-12-16:
Use the freetrack protocol. Drop me a screenshot with the protocol settings.

Tafkaa, 2016-12-16:
thanks for the reply. I am using the freetrack protocol (i think; this is all new to me). I hope this is the screenshot you are asked for:

[image]

sthalik, 2016-12-16:
Set the first checkbox to "use both".

Tafkaa, 2016-12-16:
I just tried it. Still nothing. Any other suggestions?

Tafkaa, 2016-12-16:
Nevermind. I fixed it by removing opentrack (and every registry entry i could find that has anything to do with it), rebooting and reinstalling. Everything works again now. Thanks again for the help and time !

sthalik, 2016-12-17:
For what it's worth, it started working because you started opentrack before running a game, after changing that particular protocol preference. Glad it works for you.

closed 2016-12-17 by sthalik