opentrack/opentrack #515: PS3 eye only runs at 30 FPS on Windows 8.1

ajpoinsatte, 2016-12-21:
Hi, Im having trouble getting the ps3 eye to run at more then 30 FPS with the acuro paper tracker, (i tested this issue and it seems to occur in point tracker as well ), However, with CL eye test I can get it running at 75 FPS no problem. Its only in opentrack. Anyone have any idea why? I even installed K-Lite mega codec pack and im still having the same issue. Thanks.

Len62, 2016-12-21:
Have you tried adjusting the Aruco tracker settings?

ajpoinsatte, 2016-12-21:
Yes, and it even says 60FPS in the top left corner but the video feed is definitely not 60 FPS.

sthalik, 2016-12-22:
Video feed is always 20 Hz to reduce CPU load.

closed 2016-12-22 by sthalik

ajpoinsatte, 2016-12-22:
Then why is there an option to select more then 20 FPS? its a little misleading.

sthalik, 2016-12-23:
The software logic works on full Hz. The display on the software's main window doesn't update more often than 20 Hz.

SkeleCrafter, 2017-02-26:
I agree. This is kind of misleading.

sthalik, 2017-02-26:
Even if it is, there are other priorities regarding the video feed. The CPU usage on The Octopus and video feed both takes a large part of all the CPU usage.

On the other hand, in the not-yet-released hotfix, I changed the code to perform resizing in tracker's thread, not the user interface thread. Now even with lowered preview framerate it looks more responsive. Freeing the UI thread is mostly theoretical but actually the reason for more responsive interface.