opentrack/opentrack #516: opentrack 2.3.1 release process

sthalik, 2016-12-24:
Here are the few changes so far, in random order:

- minor code quality improvements
- change to the MSVC++ 2015 compiler which is a stable release. MSVC++ 2017 RC used to miscompile the software, requiring hard process termination before it fully exited. We can also now use the official Qt installers, rather than making our own with static CRT.
- tracker/pt, filter/accela, proto/freetrack translated into Russian by Alexander Orokhovatskiy. Thanks!
- switched to dynamic CRT. **report if the executable won't start!**
- initial Dutch translation by Mathijs Groothuis. Thanks!
- tracker/pt: given correct camera FOV and clip model in real-life units, will return translation in real-world units.
- add neck displacement feature that DCS users apparently care about.
- tracker/pt: merge LED center estimation code by @DaMichel. It improves on high-quality LEDs that shine evenly over their area.
- tracker/pt: don't display oversized text and crosses on 320x240.

sthalik, 2016-12-24:
The first binary build released is  at https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa. The filename in question is `opentrack-test-20161224-2.3.1rc.7z`.

Please report if something's horribly broken, any regressions, and comment on the new features.

Tux0Topo, 2016-12-25:
When i try to open the app, it hang up the folder, and do nothing. I'm using windows 10 pro insider preview build 14986. No uac message. No anything.
Also, the window folder stop working. The only fix os to restart the explorer service.
This helps you?

sthalik, 2016-12-26:
@Tux0Topo I'd like you to elaborate on the comment you posted and deleted, thanks.

sthalik, 2016-12-26:
I uploaded the new binary and updated build name. :smirk_cat: :cat2:

sthalik, 2016-12-26:
@MathijsG if you want to translate in time for 2.3.1, drop me a note and I'll send you the .ts files.

MathijsG, 2016-12-26:
That can maybe be possible :-) please send them.

sthalik, 2016-12-26:
The stubs are at dropbox as `opentrack-2.3.1-translations-20161226.zip`.

Tux0Topo, 2016-12-26:
Sorry... i deleted the post, because the problem was SmartScreen from my windows. Sorry again...

Len62, 2016-12-28:
Latest 2.3.1 is working great for me. No issues w/PS3/IR led set-up.

MopheusDG, 2017-01-09:
Hi sthalik, can help with Spanish translation if you want. Let me know if there's a template or something and I can help with Spanish (I'm from Argentina). Thanks, keep up the great work.

MathijsG, 2017-01-09:
How do the stub works? I see many different files which isn't really handy. What's the best practice without redoing all the translations?

sthalik, 2017-01-09:
@MathijsG the stub is all the translations to be done without being actually translated. You'd need to rename the stubs to your language code, but preserving directory structure.

You can make a build environment for opentrack and `make i18n` to generate updated strings to translate.

@MorpheusDG It's all in place but I'm in no shape to help for a while, sorry.

sthalik, 2017-02-27:
🐱  🐈 Thanks to all concerned! 🍺 🍷

closed 2017-02-27 by sthalik