opentrack/opentrack #517: aruco marker idea for fully working pitch

sthalik, 2016-12-28:
Hey,

We all Aruco users known about pitch problems. I think I have a solution to get full pitch response.

If you rotate the marker 45 deg, it looks like a diamond:
<pre>
  / \
  \ /
</pre>

But that's not the solution.

Instead, **rotate the marker by 45/2** (27-ish) degrees so it's not symmetrical:

<pre>
  / \
/    |
\  /
</pre>

So there's no symmetry either up/down or left/right. Just make sure when you mount it it says +-27 degree roll. The ASCII drawing sucks.

Then tell me if you get full pitch authority without the tracker confusing itself.

sh

sthalik, 2016-12-28:
Paging @Emton, @503Kento, @Tux0Topo, @frost555. Thanks.

Tux0Topo, 2016-12-28:
I'm not sure if understood, but so far, is a no go. When trying to pitch up, the octopus start to move ackward (yaw left, pitch up). It's like the logic didn't integrate the 27 degree roll as part of the entire movement.

sthalik, 2016-12-28:
It's centering code. I can fix this in a different way. Question is, do you get full pitch authority unlike the axis-aligned marker mount?

sthalik, 2016-12-29:
To elaborate, normally you need to have a marker pitched upward cause rotating it downward would confuse the pitch value close to 0.

What camera centering mode are you using right now?

Emton, 2016-12-29:
I've tried this before thinking I could avoid the "identity pose" when looking down the problem was it just moves it to one side (and also makes the axis feel sort of funny). It's been more effective to run with high accela rotation filtering (sensitivity 4 degrees, deadzone 1.75 degrees, high smoothing 30ms, even ingame smoothing can help too I think) I tilted the shape back more a little I center the view at about 25 raw pitch, then I limit the max downward pitch angle and when I look down I simply look PAST the identity pose area.

A better solution is a separate optional filter, when the shape is at identity pose (looks like a square on the camera) then Game data pitch and yaw (the output) is Zeroed, so instead of the solver getting confused and jumping everywhere it simply holds still (on pitch and yaw axis, the others don't seem to have an issue). As the shape comes out of identity pose then the output is restored.

But with this one issue might be "how does the solver react to moving through the identity pose area?" say if you were looking back-left moving to back-right passing through identity pose, would there be a hitch or freeze? This would be some custom code to make up for Aruco deficiency but could be well worth since Aruco wasn't really made for head-tracking. I can help with any testing and feedback.

sthalik, 2016-12-29:
@Emton I can fix centering with code changes. But I need to know -- does it fix the identity pose funkiness?

Emton, 2016-12-29:
Did you want the test done with the regular positive pitch like the wiki says to have or with 0 raw pitch and just the rotation?

sthalik, 2016-12-29:
With positive and negative, especially passing through the zero point.

Emton, 2016-12-29:
The pitch pretty much works like you would expect and as long as the shape stays out of identity pose it doesn't get jumpy. However that doesn't mean the shape doesn't ever get into identity pose for instance
[image]

Also you have to consider will the rotation value be variable? Will people have to make the marker sit at 27 degrees, it can make constructing the model more tricky. 

But no matter what there will always be an area that has identity pose that's why I've been saying if you use the Identity Pose as the CENTER it should work since normally people have some deadzone in the center from curves anyway and the solver doesn't have trouble finding which way to go OUT of identity pose. Just got to stop the pitch and yaw from jumping around a simple checkbox "Stabilize Identity Pose/Center at Identity Pose"

sthalik, 2016-12-29:
But you didn't roll it +-27 deg....

Emton, 2016-12-29:
Yes I did, the numbers show up weird because of the funny angle

[image]
[image]

Emton, 2016-12-29:
But what I'm saying is the pitch "works" but all you'd be doing is moving the identity pose area, well I already use the program like that but my identity pose area is looking forward slightly down, and I just sort of avoid that spot (have other ways of looking at that spot) this angle I showed in the picture seems close to looking in my 5 o'clock low which would be even worse. Maybe it would work but it seems like the angle would have to be too precise and there's too many variables such as camera height, distance, how people wear the marker, that the compensation for the roll on the other axis would get screwy...

sthalik, 2016-12-29:
But if the 27.5 deg rotation worked, you'd have as much pitch as yaw authority. I had my own share of problems with only +-10 pitch authority before the marker stopped getting detected.

If I make a build with rotated model coordinates, it should be a big improvement, righT?

Emton, 2016-12-30:
Will it be optional? I'll help you test it if you think it can work good but you should think about what I'm saying if it might be more simple cause it's easier to make and wear a marker that is flat and level on your forehead than having specific angles, just use the identity pose as the forward center view I know it would work.

sthalik, 2016-12-30:
Of course, optional, 0, -27.5, 27.5.

sthalik, 2016-12-30:
Today's build in the usual place -- https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

Head center should work properly also.

sthalik, 2016-12-30:
Ha, the actual value of 90/4 is 22.5. Fun times.

Emton, 2016-12-30:
It doesn't work right, can't yaw right, can't pitch up, yawing left causes extreme psychosis in the octopus, can't pitch down. If I reverse the rotation option it just reverses side these things happen on.

sthalik, 2016-12-30:
Please see the same link. I think it was a typo in the trig formula.

Emton, 2016-12-30:
A lot of times I notice unwanted pitch movement when yawing with low (negative) raw pitches, for instance if I look low back right and yaw to low front left it picks up unwanted pitch movement around the 2 o'clock position. Yawing with higher raw pitches seems to work more like expected.

sthalik, 2016-12-30:
Does the red outline change shape when it messes up?

Does it improve near identity with the asymmetrical mount?

How bad are the glitches you're having in a gameplay scenario?

Is your roll around zero with that asymmetrical mount?

Emton, 2016-12-30:
I tried again and got it to stop or at least be a lot less noticeable, maybe a slightly different marker position helped. 

I tried the identity position I showed earlier and it seems out of the way (good) I can still put the shape into the identity position and it can spazz out like expected but it's not in a normal view position so the feature seems to be on the right track. 

Also the +/- for 22.5 is reversed I expect to use - since that's what the raw data is saying, yet I have to use the + 22.5 also it doesn't save the option when you go back to the Aruco settings it resets to 0 (small bug)

sthalik, 2016-12-30:
It was only a minor bug that made the combobox not restore. I reversed +-22.5.

How common are the positions where it can spaz out?

The ninja edit is uploading and will be there in just one minute.

Emton, 2016-12-30:
I got to try it out more to see about it but one thing I notice can happen easy is the shape gets lost from getting too bright as I've illustrated in this picture
[image]

Can the detection be adjusted somehow because to me it looks like there's plenty of bright/dark contrast in that picture.

sthalik, 2016-12-30:
Yes! It can actually be adjusted. I had it adjusting itself in the older versions where it broke all the time. Give it a day or few.

Why is your chin blue though?

sthalik, 2016-12-31:
I understand, it's for privacy. It's perfectly fine to crop out your face or make it nondescript like that, in case you have any doubt.

Emton, 2016-12-31:
Try to make that change soon so the shape doesn't get lost when it's bright (let it get REALLY bright) because now with 22 rotation the lighting becomes less even, yawing on one side might be brighter than the other, then the shape gets lost it fools me into thinking there's a bug.

sthalik, 2016-12-31:
@Emton I put a new build in the same dropbox directory. You may need to wait about 5 seconds as it cycles detection modes till it finds one that's for you.

sthalik, 2017-01-01:
If you have an image where it gets lost even in the last build, send it here.

Emton, 2017-01-01:
Ok I've been trying it out works pretty good so far

sthalik, 2017-01-01:
How's the 22.5 deg. pitch? Was I correct?

If it's not working too well, try rotating the camera so that the marker is close to the top or bottom of the frame, or in a corner. Or in the center. The perspective's distortion should (?) allow for distinguishing pitch near center.

sthalik, 2017-01-01:
In fact I had a user who wasn't really able to reproduce the pitch problem easily, even with neutral yaw and roll.

PS I'm having the stupidest idea of using aruco as a driving wheel for racing games.

Emton, 2017-01-01:
For the brightness modes make them go as high as possible, like in this picture looks like there's good contrast still

[image]

Now in this picture you can see it's just too bright and the shape really gets lost

[image]

Emton, 2017-01-01:
The 22.5 roll works pretty good, I can't easily put the shape into identity pose and it doesn't seem to have much effect if you do so that's good. The brightness modes are really good now you can run with 30 fps, need less extra light for the shape much more reliable.

Emton, 2017-01-01:
Hey I noticed the solver can get a little confused looking down near the identity pose, it seems to make small fast jumps earlier I had stronger accela smoothing so maybe that lessens it some.

Emton, 2017-01-01:
Another thing I notice when I roll right it adds Z for game data and when I roll left it subtracts Z for game data, yet I have Z axis disabled entirely.

sthalik, 2017-01-02:
Hey,

Try my earlier suggestion of putting the marker more into a corner of the video frame. It might give more perspective distortion for the solver to operate.

Disabling roll won't help since solver computes all rigid parameters (6DOF). If you disable it they're thrown away.

sthalik, 2017-01-02:
Drop me video frames of the last 2 overbright frames you showed please.

Emton, 2017-01-03:
Here I got it but when using ManyCam, ManyCam adjusts the image (actually makes it a lot more reliable) but I turned up the contrast in ManyCam to get the same effect. http://www.filedropper.com/brightnesscalibration

Emton, 2017-01-03:
Tell me if that video doesn't help cause that manycam is not giving the same picture when using 30 fps in opentrack I might try again later.

sthalik, 2017-01-03:
It's cause the marker takes too much of the screen. There's a minimum and maximum amount.

I won't enable asymmetric mounts unless you test it right -- that is, without absurd Accela settings to compensate for brokenness.

Emton, 2017-01-03:
Well in the pic I posted 2 days ago the marker is not too big yet it gets lost from being too bright even though there's plenty of contrast. I'm trying to get you video but manycam changes it. There's nothing absurd about the accela settings it's what works smooth for me, even if there was no identity pose limitation I still only lower the accela a little bit...

sthalik, 2017-01-03:
I need it. Please send it.

Emton, 2017-01-03:
Well this is like the 4th time you've told me I'm "doing it wrong" yet I've never seen one video of anyone else using this thing in the "correct" way or any detailed diagram about how to set it up, not to mention the whole thing is experimental anyway. I'm not sure what else I can say the "identity pose" with the -22.5 rotation now is down and to the left some (instead of straight down) but the "jumps" are somewhat smaller. 

I see the issue with the brightness modes if it adjusts for bright scenes and then it adjusts for a dark scene it can lose control moving back to the bright scene (the scene on the monitor influences the brightness of the marker a lot). Is it possible for the detection to be constant and cover all ranges? Well I got to wait till it gets dark to get another video the daytime lighting is completely different probably has something to do with the Sun.

sthalik, 2017-01-03:
If there are still jumps then my original suggestion doesn't help any.

Let's work on contrast issues. Send me either videos or images. I'll see what I can do.

Emton, 2017-01-03:
Well it might still offer some improvement you could leave it as experimental option or something I'll try to get another video later.

Emton, 2017-01-04:
Here's the best I could do with that brightness even with the right setting in OT manycam still seems to adjust the image some, here all I did was use the manycam contrast slider and increased that until the shape dissappears at 12 - 14 seconds or so into the video, in this video the brightness is the only reason the shape is disappearing. http://www.filedropper.com/brightness

sthalik, 2017-02-21:
None if this works. Let's leave things as they are.

closed 2017-02-21 by sthalik