opentrack/opentrack #518: WiiMote support: may I help ?

nicolaschaufette, 2016-12-30:
Hello guys,

I know there are request for supporting the WiiMote
It includes a IR camera that detect the SensorBar. The whole processing of the image is done by the WiiMote and the data that comes out is the position of the 1 to 4 IR leds in the image, used by the Wii to find out where the WiiMote is pointing compared to the SensorBar position (and thus the TV), how wonderful of a design it is :) 
Besides, the protocol is rather simple and there are libraries out there than support it
I know : it's very low-res camera. But what's stopping us to using two and what level of precision could we obtain triangulating 4 dots in 100Hz ? :) 

I'm a developer myself and I would be interesting to put my hands on it. My first try compilating with MinGW-W64 was not a success (but I hate Cygwin, would I fare better with MSYS ? any hope for a Wiki page ?), but I hope to have more luck with MSVC 2015

Sorry for creating an issue for what isn't, but just letting you know.

Happy New Year 2017

sthalik, 2016-12-30:
Hey,

You only need Qt 5.x. The official binary is fine. Be sure to run the `INSTALL` target in Visual Studio so that it sets up the correct directory structure.

It's great you're doing this, and thank you for letting us know.

There's a very incomplete code guide at <<https://github.com/opentrack/opentrack/wiki/Hacking-opentrack>>.

Feel free to ask questions as you run into problems.

MopheusDG, 2017-01-09:
This addon would be great ! Although I found that the Wiimote BT breaks my phone Wifi when too close and this is bad for my VR DIY headset. But to play games without headset, Wiimote with plus should be amazing. Hope you make it happen ! Thanks.

sthalik, 2017-02-21:
@nicolaschaufette using the msys2 cross-compilers is trivial

> export PATH=/mingw32/bin:"$PATH"

Then `cc`, `ld` and friends are exposed along with all the synonyms.

You'd need to download the `opentrack-depends` repo with its submodules to make a build. Also, we unconditionally require (cross-compiled, of course) Qt.

sthalik, 2018-01-04:
Response timeout

closed 2018-01-04 by sthalik