opentrack/opentrack #520: Linux build problem "undefined reference to pose_widget_impl::qInitResources_posewidget()"

surgat, 2017-01-01:
Hello! I have build problem

Versions
opentrack 2.3.0
Qt5.6.1

```
build # cmake /root/opentrack/opentrack
-- module opentrack-api
-- module opentrack-compat
-- module opentrack-csv
-- module opentrack-filter-accela
-- module opentrack-filter-ewma
-- module opentrack
-- module opentrack-logic
-- module opentrack-migration
-- module opentrack-options
-- module opentrack-pose-widget
-- module opentrack-proto-fgfs
-- module opentrack-proto-udp
-- module opentrack-qxt-mini
-- module opentrack-spline-widget
-- module opentrack-tracker-freepie-udp
-- module opentrack-tracker-udp
-- module opentrack-xplane-plugin
-- Configuring done
-- Generating done
-- Build files have been written to: /root/opentrack/build

build # make
[  1%] Built target opentrack-version
........................................................................
[ 70%] Built target opentrack-migration
[ 72%] Built target opentrack-pose-widget
[ 72%] Linking CXX executable opentrack
../pose-widget/libopentrack-pose-widget.so: undefined reference to `pose_widget_impl::qInitResources_posewidget()'
collect2: error: ld returned 1 exit status
gui/CMakeFiles/opentrack.dir/build.make:539: recipe for target 'gui/opentrack' failed
make[2]: *** [gui/opentrack] Error 1
CMakeFiles/Makefile2:826: recipe for target 'gui/CMakeFiles/opentrack.dir/all' failed
make[1]: *** [gui/CMakeFiles/opentrack.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```
How can I fix it?

Thanks!

sthalik, 2017-01-02:
I earlier removed usage of the `Q_INIT_RESOURCES(x)` macro. `make clean all` should pass cleanly.

surgat, 2017-01-02:
Works! Thanks

closed 2017-01-02 by surgat