opentrack/opentrack #522: No UI or activity after main .exe launch

Ounicron, 2017-01-08:
I currently get no response from this program when executing under Windows 10.   Currently using the portable version (2.3.0-win32-portable) on a separate logical drive from the OS. The program starts, and there is a process listed in the Task manager, but no UI launches at any point and the process shows no CPU usage at all.  I've attempted all the standard procedures, run as Admin, Win 8/7 compatibility ect. and get the same response every time.

Tux0Topo, 2017-01-08:
Check if the SmartScreen process is running in backgroun. If it's working, close it and wait a few second. If the app show, search about how to disable SmartScreen (it's easy, but don't remember how :p). If nothing happens, don't know....
Any change, please, comment.

Ounicron, 2017-01-08:
Smartscreen has been long disabled on this box.  Unless it is requisite to Opentrack somehow, I don't believe it's the issue.

sthalik, 2017-01-09:
Drop a backtrace of all threads please.

MathijsG, 2017-01-09:
Isn't it hidden in the system tray?

Ounicron, 2017-01-10:
Bizzaro occurrence: I ran it.  Same behavior as reported priorly.  I opened process explorer to get a backtrace of it, and when I fired up the process explorer it launched perfectly.  It now launches regularly every time.  I'll try to catch it in the act if it does it again, but for now I have nothing to chase down.  My apologies for the apparently false alarm.

ouoertheo, 2017-12-15:
I had this issue and I use Moonlight/Geforce Experience. When I had moonlight connected, I ran into this behavior. Going to open a new bug for that, but thought I would post here in case it helps.

sthalik, 2017-12-18:
Please do the following:

- download Sysinternals Process Explorer
- enable "lower pane view" in "view" menu
- select "dlls" in lower pane view type, "view" menu
- select the opentrack process
- locate the offending injected library

Detecting the offending library will allow for placing a warning on opentrack startup.

sthalik, 2017-12-18:
If you can't find the particular library, send screenshots with the entire .dll list. Thanks.

ouoertheo, 2018-01-02:
I've been having a bit of trouble reproducing this on demand. Seems there are some other variable in play that I'm working on figuring out. Sorry for the delay in getting back to you on this

Sent from Nine<http://www.9folders.com/>

sthalik, 2018-01-03:
opentrack deals badly with injected .dll's. This isn't something other software should do to opentrack to begin with. We already have two failure cases with "GIGABYTE OC GURU" and another similar app that make it crash on tracking start.

I'll look into refusing to load particular .dll's.

sthalik, 2018-01-03:
Please send me the Moonlight .dll name, thanks.

ouoertheo, 2018-01-06:
Here’s the dlls that load up with opentrack before I launch moonlight (works) and after (fails).

sthalik, 2018-05-14:
It's now in FAQ, see wiki.

closed 2018-05-14 by sthalik