opentrack/opentrack #523: Mouse output is sticky on small movements

Kissamies, 2017-01-08:
I've been lazy getting new versions because I was happy with 2.3-rc21p11, but decided to update to 2.3.0 today. Unfortunately it seems like the mouse emulation output doesn't seem to work as well in this version as with the one I used before. It gets off-centered much easier, not returning the same position when I center my head. The earlier version was a bit drifty too, but not nearly as bad. I think the issue might be how it doesn't seem to register the movement when I turn my head very slowly. The input does and the "game data" box as well, but somehow is isn't translated to the mouse unless I turn it faster. Feels very sticky.

I wish I could use freetrack or even joystick output (though I wouldn't need to do that with EDTracker), but unfortunately I have to use the mouse workaround since Star Citizen is yet to fix its support for either of those. Mouse output is useful for old games too. Well, I suppose I can keep using the older version for now.

soryy708, 2017-01-14:
Sounds like something is an integer when it should be a float, or some conversion error. Needs investigation.

sthalik, 2017-01-14:
It's likely your filter settings. Ctrl+v the filter window contents here, please.

@soryy708 things are kept as a real internally and only converted to discrete when Windows expects it.

sthalik, 2017-01-14:
@soryy708  As for mouse API we're using -- we're submitting a relative mouse movement without needing to read its previous position. As such, there's no race condition that would occur in the read-then-write scenario.

sthalik, 2017-01-14:
If it drifts with time a lot, it may also have lots to do with low framerates.

Kissamies, 2017-01-14:
I myself suspected it might have something to do with the filter so I tried it with no filter and it still happens. Anyways, this is the latest filter I tried.

[image]

Framerate being the cause is possible because Star Citizen has pretty bad FPS in PU, but the mouse movement being sticky happens even then I'm just testing it on desktop.

Hmm, come to think of it, I haven't been playing any other head tracking games lately. I should check if I have issues elsewhere. I'll try that and report.

sthalik, 2017-01-15:
Disable mouse acceleration for the duration of playing SC. See if it still drifts.

Kissamies, 2017-01-15:
I have both the desktop and game's mouse acceleration disabled. I don't think it's it. Remember that I don't have this drifting problem with the older version. I really think it's the sticky movement, maybe in the mouse protocol, that causes the drift. On desktop, I can make the cursor move close to the edge by turning my head fast and then come less than halfway back to the center by turning my head back to the center at slower rate.

sthalik, 2017-01-16:
@Kissamies I found something dubious in the code. See today's build on <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>> if it helps any.

I ran the test tracker with no filter on the smallest sensitivity setting, and it made a really slow circle in Skyrim SE.

Sorry for not believing you at first! We get plenty of bug reports that turn out to be configuration issues.

Kissamies, 2017-01-16:
Looks promising, though I can't say for sure yet because I made myself a cludge of a 3 point model with some IR leds and bendy straws and have been spending this evening playing with that. Suffice to say I haven't configured that well enough to get reliable results. I'll reconnect my EDTracker tomorrow for some proper testing.

Don't worry about the not believing me at first thing. I have a lot of patience. One has to be to deal with the workarounds to get head tracking work in SC.

sthalik, 2017-01-17:
Hey,

You can compare the old and new versions using the test tracker. Just keep in mind that you have a sharp discontinuity on -180->180 in the old tracker. That's an unrelated bug.

sthalik, 2017-01-17:
@Kissamies take your time and confirm positive or negative once you have your preferred device. Thanks.

Kissamies, 2017-01-17:
I think I have this tested out now. There is slight drift, but I estimate it's not worse than the 2.3r21p11 I used previously. Hard to quantify it precisely with the center and zero not working on that build. Could be that it's not the protocol's fault. Could be my tracking device, or mouse just always drifring a bit. All in all, very good. Could have pretty long dogfights without having to adjust the mouse manually. I'm happy with that.

Noticed a minor issue. When toggling while the in motion, the mouse seems to inherit a bit of the velocity and keeps moving ever so slowly for a while. A video to demonstrate:
https://www.youtube.com/watch?v=19vuLlPiVkY
Doesn't have to to be that fast, any movement seems to cause it. 2.3.0 doesn't do that.

With EDTracker the workaround is using "Toggle tracking" instead of "Toggle", but PointTracker takes a bit of time to "boot up" so toggling it on and off is bit inconvenient. In fact 2.3r21p11 has this issue too, but I never noticed because I used the tracking toggle.

Also an idea: It would be nice for mouse emulation to start the tracker toggled off. Star tracker, navigate OS to start the game, navigate the game menus to get playing and then toggle it on. Of course one can put it on hold manually after starting it so it's not that important feature, but it just crossed my mind.

sthalik, 2017-01-18:
@Kissamies even with toggle, tracker and filter data are processed. Only that they're not passed to the protocol.

Presently we have no infrastructure for a tracker to toggle off the emulation process. The modules are very generic.

You can also use the "restart tracker" button and friends in place of toggling.

sthalik, 2017-03-20:
I think it's fixed.

closed 2017-03-20 by sthalik