opentrack/opentrack #524: New Feature: Multiple Keys/Joystick bind to shortcuts

MopheusDG, 2017-01-08:
I think it could be nice to have the possibility to map more than 1 key or joystick input for the same function. For example, have C key to center view and also Joystick Button 10 at the same time, this way it's easy to center when you switch between a game with keyboard (Arma) and a game with Joystick or Wheel (Live for Speed, Etc).

Happens to me all the time having to switch bind keys when switching games.... I use the same profile for all of them but key binds are the only thing I need to change for confort.

This probably is quite easy to add and I think could be useful.

Thanks, Opentrack is amazing.

MathijsG, 2017-01-09:
Why not use a profile per game? That's how it's supposed to work. You can even bind a game to a profile so you don't need to manually change the profile.

MopheusDG, 2017-01-09:
In my case I don't change any settings between games at all, so I only have 1 profile for all my games. It's easier to use this way.... I don't know if some other games that I haven't test need more detailed settings, but usually the only setting I change is between mouse emulation and TrackIR emulation... and I couldn't find a way to switch those with bind keys (maybe switch profiles is a good idea for that), but again, if I play Alien Isolation for example, sometimes I use keys and sometimes joystick, so will have to make 2 profiles for the same game just because I don't have multiple bind option ?

Anyway, It was just an idea from my experience with the app. Maybe is not that easy to do, don't know... maybe as you say, it's not practical... although for me it would. Just my 2 cents :)

MathijsG, 2017-01-09:
I have it the other way around, I have two profiles per game, because sometimes I use tracking clip for headphones, and sometimes I use the tracking hat. But maybe it's a good option to allow multiple buttons for a certain function.

MopheusDG, 2017-01-09:
And how do you bind the game to 2 profiles ? is it possible ? Or do you change manually ?

MathijsG, 2017-01-09:
I don't use the automatic functionality, I think it isn't possible to do that automatically.

MopheusDG, 2017-01-09:
Then multiple bind could be good, maybe also some additional shortcuts could be of use, but not sure which one's. For me Key and Joystick options for centering and toogle on/off is enough, but some could use more.

sthalik, 2017-01-09:
It's hella annoying, yeah. :+1:

sthalik, 2017-02-18:
Done. Please report issues.

closed 2017-02-27 by sthalik