opentrack/opentrack #525: Android: Raw Data keeps on turning although my head/phone doesn't move at all

SevenSiSix, 2017-01-09:
Good day, thanks for making this possible, it is amazing! I only have this little problem where the yaw keeps on increasing/decreasing even when I already stopped my head from moving. I don't know if this is a problem with my phone or something else. I am using a OnePlus 3.

Thanks in advance!

sthalik, 2017-01-22:
Move away from any magnets. Eliminate the possibility of magnetometer error influence by putting headphones and speakers out of the room temporarily and see if it's still broken like that.

Also how many degrees are we talking about? 60? 30? 5? 2?

SevenSiSix, 2017-01-23:
Thanks for your response, I've tested it with my speakers close by, or completely removed from my room. And that doesn't change it. It could be because it had a good day, it's right now about 2 to 5 degrees at the max. But I've had it happen with up to 30 degrees of rotation.

Is there an easy way of changing the source on my phone. I think most phones have multiple ways of tracking the direction of the phone. Maybe we could then choose the source we think works the best.

sthalik, 2017-01-24:
@SevenSiSix android has one API to get rotation from phones with a gyro. We're using that one API.

sthalik, 2017-01-24:
We COULD be using gyro+acc AHRS but Java is slow.

SevenSiSix, 2017-01-24:
I understand, thanks for your help!

closed 2017-03-01 by sthalik