opentrack/opentrack #526: Odwrócona Oś Y  w OT 2.3

MZielak, 2017-01-11:
Witam, przeglądałem pobierznie wszystkie tematy ale nie widzę mojego problemu. Kamerka PS3 w Eye -test obraz prawidłowy natomiast w OpenTrack (sprawdzałem od 2.3rc98 do 2.3.0) oś Y jest odwrócona. Wygląda to tak że posiadając 3 diodowy cap trójkąt z nich tworzony jest do góry nogami i przez to wszelkie ruchy są zakłócone. Nie pomagają żadne inwersje w ustawieniach OT. Testowane w FTNoIR daje ten sam efekt. Mój system to Win10 64-bit (upgrade z 8.1 64-bit) Pozdrawiam

sthalik, 2017-01-11:
Trójkąt jest okej, masz model zrobiony na odwrót. Użyj "custom model" to ci zadziała jak trafisz odpowiednio w plusy i minusy.

MZielak, 2017-01-11:
Dzięki za szybką odpowiedź - czy dobrze rozumiem że mam źle wykonany model cap-a? dlaczego więc w EYE-test wszystko jest jak trzeba? Gdzie znaleźć ten "custom model" ? przepraszam ale jestem jeszcze trochę zielony w tych kwestiach.

sthalik, 2017-01-11:
Tam gdzie masz zakładki jest ich 3, cap, clip i custom.

Ale jest proste rozwiązanie -- w configu masz wysokość i zmień ją ręcznie na ujemną. W sekcji `tracker-pt`. Ale jak zmienisz opcje PT to znowu będziesz musiał zmieniać ręcznie.

Model masz ok tylko masz górę z dołem na odwrót wykonane.

sthalik, 2017-01-11:
Ale powiedz mi, czy renderuje się odwrotnie obraz w opentracku w porównaniu z cl-eye test? Bo to jest bardziej podstawowy problem.

MZielak, 2017-01-11:
...no właśnie o to chodzi że w cl-eye test wszystko jest jak trzeba - czyli trójkąt z diod osadzonych na czapce ma takie samo odbicie na monitorze a gdy odpalam OT jest na odwrót czyli do góry nogami - trójkąt w okienku OT jest diodą środkową na dole a dwie boczne z daszku są u góry...

sthalik, 2017-01-11:
To jest bug z niepoprawnym użyciem kodeków w opentracku. Kilka osób miało ten sam problem co ty. Doraźnie, uruchom manycam czytające obraz z kamerki.

Jest otwarty któryś bug dot. nowej implementacji obsługi kamerek. Zobacz na "experimental builds" z wiki `amcap`, to jest ta nowa obsługa. A w bugu jest opisane czemu tyle to zajmuje. Jakby mogło być gotowe w ciągu dnia-dwóch to by było już dawno :(

Duplicate of #457.

closed 2017-01-11 by sthalik