opentrack/opentrack #529: More than one aruco marker

sam-ogon, 2017-01-15:
Hello

Is it possible to implement more than one aruco marker tracking? One is good when using monitor, but not enough for HMD, since it doesnt work above 90 degrees left or right turn.

The solution can be adding two markers on sides of the HMD, and perhaps fourth at the back. I know that openCV supports multiple markers recognition, but dont know how OpenTrack handles this. Perhaps this would be easy.

This is what i'm talking about [HMD with multiple markers](http://cdn.uploadvr.com/wp-content/uploads/2015/09/StarVR-eyetracking-vr-5k-210-fov.jpg)

sthalik, 2017-01-16:
It's possible, yes. But the program has a heavy work backlog and I can only do so much. If someone sends a pull request with the implementation, I'll review it.

I tried using opencv's aruco contrib library but the original aruco library works better. It also allows for detecting multiple markers with different id's.

For a tracker like this, at minimum one should be able to specify multiple marker id's, position and rotation for each, and handle this logic for arriving camera frames. It's not hard to write and it's very easy to write opentrack modules. See "hacking opentrack" on the wiki.

Tux0Topo, 2017-01-17:
Just an idea... Putting some lights inside a paper cube with markers in it, and playing with brightness and contrast, the detection gets better... I've tested that... ;)

sam-ogon, 2017-01-17:
@Tux0Topo :  This is not about detection problems, but lack of the marker visibility when you turn your head back. This only applies to VR goggles/HMD's.

@sthalik :  Unfortunately my coding skills are virtually nonexistent, so I don't understand much of what's happening inside the source files. There is no chance I'd be able to write such module by myself. Perhaps there is a way I could help in creating it?

sthalik, 2017-01-18:
@sam-ogon if you know someone who wants it and can write it, direct that person here.

And to be clear, this is a jolly good feature and all, but we have a serious backlog and getting it within few months is the ETA here.

leCaTo, 2017-01-27:
Hi, i was working in a way to achieve this. I actually just found out about opentrack and seems great!

My ultimate goal was a multi-camera set up that allowed 6DOF tracking in a "extended play area" using fiducial markers. All this in a way that allow playing VR games as it would an "VR headset + external tracking" system but using a low-cost headset (phone+gcardboard-like).

After a little research i figured out that RiftCat's VRidge has the game streaming covered along with the steamVR interface (enabling playing most games), and they talk FreeTrack 2 protocol !!

Correct me if i'm wrong, but it seems that a proper opentrack module might feed the required 6-DOF pose. I work on image processing and have experience on multi-camera calibration/localization methods.

¿All this sounds appealing to you? tell me if this is too far out of the project's scope. I've played around with opentrack, quick-scrolling the source code and a few things came to my mind already.

Tux0Topo, 2017-01-28:
PointTracker AND Aruco tracking options, allow 6DOF pose. The big problem that they have are related to limited range of rotations.
Also, Opentrack and Freetrack only allow 1 camera input.
So, if using 1 camera is a limitation that can't be fixed, using multiple markers for multiple faces of the object is a decent solution. If using multiple cameras is an option, using 1 good marker per object can do most of the job. Using multiple cameras, and a few markers per object should be able to do the best job.

sthalik, 2017-01-28:
But you don't even need multiple cameras. There can be multiple markers each with different coordinates.

There's the "aruco board" functionality and it can aid with pose estimation -- probably, I never gave it a chance. But please first try to specify several markers around the VR headset along with their id's. There's not much code to write. We're doing half of that work already.

sthalik, 2017-01-28:
@leCaTo that is correct, you can detect multiple markers and get their id's as well. The current Aruco module isn't written very clearly but it's bearable.

Multiple people asked for that. If you wrote that, please submit for review and work with me to get it in tree. It's painless. As usual, our policy of crediting contributors applies.

leCaTo, 2017-01-28:
Fair enough, will start towards multi-markered object tracking using a single camera. Although I've made some testing and seems to me that to achieve robust tracking in an area of a few meters you need multi camera and big markers (around 25cm).

Some questions and concerns:
- ARToolkit supports already multi-marker object tracking and they established a simple configuration file format. At the end of: http://artoolkit.org/documentation/doku.php?id=3_Marker_Training:marker_multi

You would prefer a module based on aruco, right? I can implement a parser for this kind of configuration format or just make an artoolkit based module. I've read the aruco paper and seems very promising, but artoolkit has several features and utilities that may be beneficial. For example, do you know how well aruco performs in terms of adaptive thresholding?

- Seems to me that opentrack do not ask for the camera's calibration, image rectification may improve aruco performance. I do not have much experience on programming ui with qt, but an interactive camera calibration ui may be the way to go. This kind of feature would be beneficial for every other type of tracking algorithms, right? I'm not familiar on how PointTracker works, but image distortion correction may be helpful.

sthalik, 2017-01-29:
@leCaTo it needn't be based on aruco. Especially given that `tracker/aruco` code is bad and you seem to have a working solution.  `tracker/test` is a stub for a basic tracker, and you can grab the video widget logic from the "opentrack-cv" module.

If some of your libraries depend on opencv, please try using our forked version. See other repositories for the opentrack project.

I calibrated my PS3 Eye with the opencv chessboard pattern ages ago and it wasn't much help. Also the coefficients were radically different each time.

leCaTo, 2017-02-16:
@sthalik understood, i may have time to look into it next month. Concerning calibration, distortion coefficients are pretty hard to judge as two different set of values could both be good approximated solutions. Sometimes the camera actually doesn't have a determinate type of distortion and some coefficients may result superfluous and numerically vary. I'll do some testing, but i have always worked with calibrated cameras while doing marker recognition.

sthalik, 2017-02-17:
Hey,

Most of the time we had problems like that it's the fov -> focal length formula being incorrect. PT has a correct formula for reference.

closed 2018-05-14 by sthalik