opentrack/opentrack #530: DCS

Zulfiqarib, 2017-01-16:
Hi,

 I have been using opentrack since quite a few months , my issues is DCS 1.5.5 , when i fire up DCS the tracker is working but nothing happens in DCS. so I reboot the system and then it may work or it not. 

 I also use XPlane 10 / 11 , if opentrack does not work in DCS then i fire up Xplane 10 and then DCS and it works.

Please help

Rgds.

LennyPenny, 2017-01-16:
Try to choose the trackir protocol in the settings instead of what's currently selected

Zulfiqarib, 2017-01-16:
@LennyPenny 

 I have already tried using the Trackir protocol but no go. Though i will try it again.

sthalik, 2017-01-16:
@Zulfiqarib give me a few days please. Then I'll send you a debug version of the NPClient library. It's been fucking hectic around here since Saturday...

Zulfiqarib, 2017-01-16:
@sthalik 

No worries , will be great to have it rectified.

Thanks sthalik

sthalik, 2017-01-16:
@Zulfiqarib when you start DCS and it works with opentrack, does the title say "DCS: Black Shark" or something else?

sthalik, 2017-01-16:
@Zulfiqarib also please confirm that if you start the game as admin, you start opentrack as admin. But if you don't start the game as admin, opentrack doesn't start as admin either.

sthalik, 2017-01-16:
@Zulfiqarib also please say what the title says when you start DCS and it doesn't work.

petsild, 2017-01-16:
Hi,
me OT works in DCS 1.5.5 as always great, do check the integrity of files DCS_Updater.exe repair instructions here:  https://forums.eagle.ru/showthread.php?t=114030

Then check the settings of the axis trackir in DCS.
[image]

Zulfiqarib, 2017-01-16:
@sthalik 

Hi,

 When it works in DCS the title is Black Shark , but it does not work it will have no title for the game.

A few minutes back I started DCS ran opentrack and it did not work. So I started Xplane 10 and opentrack was tracking, then i quit Xplane 10 let opentrack be running and ran DCS and now the title says Black Shark

sthalik, 2017-01-22:
@Zulfiqarib are you ever running two sims at once? Always be sure to quit the first sim fully before starting the other one. I'm not sure if that's the case on your end.

Zulfiqarib, 2017-01-23:
@sthalik 

Only one sim at a time, i use a laptop and not possible to run two sims together , lol, you expect me to fly two aircraft at once :) 

I always quit one b4 starting the other.

Zulfiqarib, 2017-01-28:
@sthalik

The only way i can use opentrack with DCS is to start opentrack before starting DCS.

sthalik, 2017-01-28:
So take this debug build <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>>

The file is named `opentrack-test-20170128-test-freetrack-proto.7z`

You should get lines in the form of

> game-id 1901 game-name "Falcon BMS" table empty

Try not to close opentrack via the console window as it leads to unclean termination.

I have "freetrack disabled, trackir enabled" in the protocol settings. I should rename the latter to "NPClient enabled/disabled" but people would hate it.

Zulfiqarib, 2017-01-28:
@sthalik 

 Thanks will try it in a few hours. I am on windows 10 btw

Zulfiqarib, 2017-01-28:
@sthalik 

Thank you, seems like it is working, It said Black Shark.

One another , i think it seems to be heavy on the CPU ?  I am on a laptop and noticed that the CPU load is higher when i ran this opentrack. 

Will check with Xplane a bit later and report back.

BTW can you help me here too ? I look to the right the everything zooms in and blocks the view, e.g. when you are in an aircraft and look right , here the seat zooms in. I have tried everything but it does not help. I am using 3 point IR clip.

sthalik, 2017-01-29:
@Zulfiqarib I didn't make changes. I only added the console window for debugging.

See at the camera feed how your clip looks like when rotated.

Zulfiqarib, 2017-01-29:
@sthalik 
 
 That right look is the same as the old version but no worries i will check it.

Do you want any logs or something from the test version ? I wouldn't know how to give you the logs.

sthalik, 2017-01-29:
@Zulfiqarib just describe what happens in the console window once you switch to DCS and it doesn't work.

Zulfiqarib, 2017-01-29:
@sthalik 

Probably today may or may not be possible to , but if i am able to will post the console logs. 

Thanks.

sthalik, 2017-01-29:
👍

Zulfiqarib, 2017-01-29:
@sthalik 

I tried it again and it did not work, there were no error messages , so quit DCS and test version opentrack and then start test version opentrack and then DCS and then it worked.

Let me try it a few times and see if any errors do pop up.

I do have some more queries though, it's all got to do with the setup that's all.

sthalik, 2017-01-29:
@Zulfiqarib you had no game switch messages with DCS at all?

Zulfiqarib, 2017-01-29:
@sthalik 

I will check it again , give me a day,

Zulfiqarib, 2017-02-02:
@sthalik 

If I load DCS and then opentrack it does not work and there is no error in the console showing up. I always need to run opentrack before running DCS. I mean it's fine if its that but if it can be fixed that would be great

sthalik, 2017-02-02:
We only have limited means to communicate with DCS. We're doing what can be done but that's not enough apparently.

Zulfiqarib, 2017-02-02:
@sthalik 

Whenever you can fix it no worries, i can always use it by running it before starting DCS.

sthalik, 2017-02-02:
@Zulfiqarib it's not a matter of "when" but communication with the DCS executable being limited.

What happens if you:

- start tracking
- start DCS, it works
- stop DCS and tracking

- start DCS
- start tracking

does it show "DCS: Black Shark" like it does when it works?

Zulfiqarib, 2017-02-03:
@sthalik 

 If I run opentrack before starting DCS then it does not matter how many times I quit and start DCS it will work.  But it won't work if I start opentrack after running DCS.

albertolg, 2017-02-03:
@Zulfiqarib so where is the problem? start opentrack then DCS or any game you want to play, I believe it's DCS that don't recognize the trackir so it doesn't enable it and not an opentrack problem

Zulfiqarib, 2017-02-03:
@albertolg 

 I don't know , but if it can't be fixed then it's ok. I can always start opentrack before running DCS.

 It works fine with XPlane.

sthalik, 2017-02-04:
@Zulfiqarib I'm leaning on calling it a DCS bug but if we did things a bit differently then it'd work even with opentrack disabled.

@albertolg normally opentrack can be started at any point, including not at all.

Zulfiqarib, 2017-02-05:
@sthalik 

THanks for all the help , i will run it the way it is.

closed 2017-03-01 by sthalik