opentrack/opentrack #532: Can tracker-rs be used with newer RealSense SDK?

ansorg, 2017-01-16:
Seems like Intel updated and restructured the SDK in the new version https://communities.intel.com/community/tech/realsense/blog/2016/12/08/whats-new-in-the-intel-realsense-sdk-2016-r3 

I wonder which of the individual pieces would be the one I need to install to use SR300 with opentrack?

ph0b, 2017-01-16:
Hi,

you can't switch to the new release just as a user, we need to integrate the new SDK first (basically: download the new SDK core and face modules, recompile the tracker, and update the bundled runtime installer). I'm not sure there is any improvements on head tracking in this release so it's not my first priority to do this. The size of the face runtime is still smaller once installed, that's already a welcome improvement, so I'll do it anyway.

btw the new SDK structure allows to directly redistribute the runtime without using an installer, I think I'm still going to integrate it like previous releases since the runtime is quite big and I don't want to cripple the build for people not using RealSense.

sthalik, 2017-01-16:
@ph0b can you take the code changes from here?

sthalik, 2017-01-16:
@ph0b it appears that there were no API changes. Please take it from here or point to redist files at least.

ansorg, 2017-01-18:
@sthalik finally found the R2 redist. thanks

ph0b, 2017-03-30:
The RealSense SDK Runtime 2016 R2 redist installer is in opentrack installation directory, under doc/contrib. 

The "install runtime" button in the module configuration and error messages should trigger it but the path is wrong in current release.

severgun, 2017-07-21:
I can't install SDK. Integrity check failed
[https://image.prntscr.com/image/IJMHl7BvSaSHNdB5_Wyxxg.png](url)

ph0b, 2017-07-21:
I've just verified the installer - removed and reinstalled everything without encountering such issue.
Can you please try again? the download may have failed because of a temporary issue with the server or with your internet connection.