opentrack/opentrack #533: When the phone input moves from 180 to -180 the game data does a reversed 360 rather than just keep turning

Ounicron, 2017-01-21:
Using:
Windows 10, Opentrack 2.3.0
Nexus 6p with either Headtrack or FreePIE IMU (behavior is consistent across both input software)

When setting to mouse emulation for output, the software does a long way around approach to the yaw or pitch hitting the maxima of the output from the phone.  When the Raw Tracker makes the jump from -180 to 180 or vice versa the game data output simply spins 360 in the counter direction to get to the other side. For example if you're at 180 and turn another degree, the Raw Data jumps from 180 to -180.  The Game data however simply counts smoothly (but very rapidly) down from 180 to -180, resulting in a flat spin in game.  It will do likewise in reverse creating this wrap line that the game data refuses to cross.  I have seen other examples of this in the issue tracker, but none with a solution that I can implement.  If there is already an existing solution for this that I am unaware of, please advise.

sthalik, 2017-01-22:
Go to "experimental builds" on our wiki here, download the latest build.  So far there's only been one build this year. Be sure to disable the filter. You shouldn't reproduce it this way.

Ounicron, 2017-01-22:
Confirmed it no longer does the reverse spin.  I do now have other artifacts, but as this is a pre-release build I'm not sure what should be working and what not.  Let me know if you desire further information.

sthalik, 2017-01-22:
It's expected to work fully. The .apk has been here for a long while now.

Ounicron, 2017-01-22:
I updated the .apk on the nexus and now it is functioning normally, there's a light skip of 5 or so when it crosses the GMT as it were, but it is significantly better.  Thank you for the updates.

sthalik, 2017-01-22:
The skipping on unit orientation isn't known to me. Feel free to comment further if you want to look into that. For now I'm marking as fixed.

commit 8583d335 (2017-03-17): filter/accela: special-case the -180/180 rotation boundary

Previously crossing the 180 degree boundary got treated as a
large rotation. This is of course incorrect.

The error in fact lies only in the filter. Special-case the sign
changes.

We're centering prior to filtering in logic/tracker.cpp so it's
all right in the world.

Issue: #533
Reported-by: @Ounicron
Fixes: #533
closed 2017-03-17 by sthalik