opentrack/opentrack #534: one little wish o:) "zero while not held" hotkey

mm1k3, 2017-01-23:
hello guys :)

first, i wanna thank you for developing and maintaining our lovely opentrack. i am using it for couple of years with ed tracker and it works flawlessly. since i am just a boring gamer, i am using it to play elite dangerous and rust, and it brings a lot of immersion to my time consuming hobby.

the reason why am i bothering you is rust. although opentrack works perfectly, there could be one little adjustment to hotkeys section. since rust is competitive FPS, it is not that easy to shoot something while tracker is working so it would be nice if we could have option that tracker is by default at "zero position" (for moving, shooting, crafting etc), but if we hold hotkey (in my case, perfect one would be SHIFT) then tracker software reads positioning and translate it to proper value based by curve (or line) formula, for looking around. it would be something like "zero while not held", and could be switched on or off in options (with setting of hotkey).
also, adding option to assign just CTRL, or just SHIFT or just ALT key as hotkey, would be great addition for all "WASD" fps players, specially for this hotkey.

once again, i am sorry for wasting your time. i am sure that you guys have something more important to do, but if you find some spare time, please help o:)

best regards!

--miljan

sthalik, 2017-02-21:
Unfortunately the request is too specialized. I'd like to add scripting support in the future but it's a lot of work to expose all functionality users may want to script.

Until then you may find some workaround, like a keyboard-scripting program that keeps a key pressed unless another key is pressed.

I won't support binding modifier keys themselves. For the time being you may find some aforementioned input remapping program.

sthalik, 2017-02-21:
See #492 for the overall description of scripting opentrack's internals. I'll work on it when it's time.

closed 2017-02-21 by sthalik

mm1k3, 2017-02-22:
"too specialized"?

guys, i am suggesting you how we could use head tracker in fps. it's not necessary to be superb smart to understand that this technology, although is available for more than a decade, is used almost exclusively in cockpit simulations. it is very sad that even developers of head tracking software are not eager to push head tracking devices further into gaming.

making head tracker better for use in fps is not too specialized request. you guys have a serious lack of vision :(

sthalik, 2017-02-22:
@mm1k3 it's not the project's job to "push headtracking/VR support" to other genres. It's game developers' chocie. Furthermore, typical FPS games don't distinguish between head position and body position, Arma is the sole outlier here. In other games you can only control the mouselook, i.e. aiming, clicking in menus, everything a mouse does. This isn't something the software will strive to support.

mm1k3, 2017-02-22:
arma and rust. and both of them have less than 1% of players really using head tracking. probably less than 0,001%.
i know it's not your job to "push headtracking/VR support" to other genres, but you guys could try to make software more useful. without your support, i am sure that developers will not give a floop too.
fps games are not cockpit simulations, and therefore we need different approach when implementing head tracking to this kind of games. my suggestion is just an attempt to make things better for fps gaming and head tracking devices.

sthalik, 2017-02-22:
We can't support headtracking properly if mouse is the only interface to move the player character's head. Furthermore we can't do it if user's body and user's head rotation can't be disconnected in game.

Head tracking is actually quite popular in Arma, see Krause from UO for instance. He also plays Falcon BMS. I used it myself and it's a decent experience for situational awareness.

We'll eventually support it via Python scripting but it's a lot of work to expose any opentrack innard a user might want to interface with. It's low priority since I have Vive and EyeX to finish first!

The scripting support's definitely something given you're asking for a completely unsupported (and in my opinion, not feasible) use-case.

Besides, if you can't keep it enabled, it already implies something's wrong with the use-case.

mm1k3, 2017-02-22:
:)))))

you should work as a PR for government. the way you are avoiding real arguments and focusing on what you think is important is epic.

scripting, scripting, scripting... i could probably write the whole software too, probably better than you 
do, but that's not the point. the point is how to make your software useful without scripts, and third party software.

i am not discussing about OTHER fps games, i am telling you how you could implement better head tracker support in EXISTING GAMES. rust and arma HAVE separated controls for moving whole body and moving just head. if you guys have no idea how to make head tracker useful (but not just when driving helicopter in arma), I HAVE, and i suggested it.

and what is your answer?
it's not so popular to use head tracking device in fps, and therefore, you request is "too specialized".

lack of vision my friend, lack of vision.

sthalik, 2017-02-22:
No shit. I'm actually giving all the technical reasons with no PR lousy shitheads to turn them into buzzwords.

Go to #492 and see the types of requests people have for the software. If I were to satisfy all the requests for custom keybindings, we'd have three tabs worth and counting.

Also see <<https://github.com/opentrack/opentrack/issues?q=label%3Awontfix+is%3Aclosed>> for kinds of stuff people request. The software has a clear direction, it's not 3ds Max or some Outlook.

Since Rust has, as you say, view disconnected from body orientation, you'd get it working comfortably with nonlinear yaw/pitch mappings. Spend a decent amount on turning up to 30-45 deg yaw, then a higher slope for up to 90, then a very sharp slope up to 180-ish. That'll do.

Also note that Accela has capabilities for a smooth, non-jerky view for more comfortable play. Put smoothing up to five milliseconds, increase sensitivity somewhat, but leave deadzone at recommended `0.03`.

With that setup I don't see an issue where head tracking's situational awareness would break stuff when aiming through iron sights or hipfire. It never did for me with Arma even when using zoom. And I used way more sensitive mappings.

You have the toggle button. There's also the "while held" button functionality inverse to your request. You'd ask for a third variant? We'd reach a combinatorial explosion soon.

I don't see a problem with bundling scripts with the software (when scripting support is done, of course), allowing them to register keybindings and dialogs. But for the core functionality, the software needs a direction to avoid growing into a larger blob with each random request. There's probably more rejected requests that weren't labelled since the "wontfix" label is rather new.

> scripting, scripting, scripting... i could probably write the whole software too, probably better than you do, but that's not the point. the point is how to make your software useful without scripts, and third party software.

That's sarcasm, but here's a piece of trivia -- opentrack's user interface contains no program logic on its own, other than what's directly necessary to display stuff. It's entirely possible to add another user interface with keyboard and joystick listen support, Now come to think of it, it can easily be done with Python 3, `cffi` and PyQt 5. I don't imply this is a valid workaround, of course.

sthalik, 2017-02-22:
See [combinatorial explosion](https://en.wikipedia.org/wiki/Combinatorial_explosion) on Wikipedia.

sthalik, 2017-02-22:
And you clearly don't expect me to check all random keybindings from the requests after the changes, do you? Given how niche they are, I'd expect them to stay broken across several releases in a row.

Finally, I provided you with a solution for usage in FPS like Arma and Rust. They're a supported use case unlike what you claim. The solution is different than your own suggestion but that's that.

mm1k3, 2017-02-22:
mate,
i am playing games with head tracker, and i know why it is linear 1:3. not because i am lazy to configure "a curve", but because it is the only way my neck and my head are not getting too tired after few hours of gameplay. every curve is making head moves to intensive and unnatural, and it harms more then it benefits. if you are using headtraker for gaming, not just for testing, you now that too.

it is not really a problem for me to make a descent macro (and i have it, already), that will do what i have requested here, but making head tracker in fps hard to configure is bad for you guys, and bad for headtrack gaming too. if it's not "plug and play" most of the users will not use it.

why are you so stubborn? lazy? nooo, i doubt. i really do believe that you are one of guys that kept head tracker in dungeon of cockpit simulators for all these years. you guys just don't see further than your nose, and in your head - this device is simulation only device.

well, it is not. and it could be WIDELY used in FPS games too, it's just necessary to adopt the way of using software to that kind of games. if you don't get it now, you will get it AFTER your competition gets it before you.

"shitheads to turn them into buzzwords" :)))) you lack some home education too, but this is not place for your personal development.

over and out.

sthalik, 2017-02-22:
You crossed the line with the insults. Just take "no" for an answer without such fuss. I've elaborated my standpoint clearly enough. I won't elaborate on any more technical points given you started to throw around personal attacks.

sthalik out.