opentrack/opentrack #535: REQ: Bigger resolution for curves

albertolg, 2017-01-24:
Hi, would be possible to have a bigger resolution for the horizontal value of the axis curve? Right now mine are this, but it's very difficult to obtain a smooth transition, I'm using the edtracker DIY

[image]
[image]

albertolg, 2017-01-26:
forget it, modified the scaling to 2x in the edtracker GUI and now is much better

albertolg, 2017-01-26:
well, I spoke too early... I have too many spikes with 2x so hope this can be implemented thanks

Kissamies, 2017-01-26:
While an idea worth considering in general sense, can't you use EDTracker UI to crank up the response scaling to get more movement out of your device?

LennyPenny, 2017-01-26:
@Kissamies the problem is actually the reverse, the edtracker seems to really sensitive which makes editing the curves at this scale so hard

albertolg, 2017-01-27:
well, I adjusetd the scaling to 2x and worked well, then when I tried later it seemed too jumpy so I reverted back to 1x. I have mapped the roll input to the X axis, I will do more tests with different scaling value

sthalik, 2017-01-27:
The idea here is that the mapping window wouldn't fit on a smaller display's resolution. We have to take care of the lowest common denominator here.

For users of non-180 deg devices -- pretty much everything non-VR -- we could clamp the mapping graph to the user's chosen value. I have some flu and it's gonna take a few days just to get strong again. I'm also burned out from sole-programmer duties. Can someone do it and submit a pull request please? It's actually really simple.

sthalik, 2017-01-27:
And to be clear, I'm gonna introduce clamping the mapping to a chosen value. It's been annoying since basically forever. So it's approved.

sthalik, 2017-01-29:
So test builds are here as usual <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>>

And the filename is `opentrack-test-20170129_01-zoomed-mappings.7z`

Also please don't call them curves 🐈

sthalik, 2017-01-29:
Please note that if you have a point that gets cut out by zoom, it won't take effect when zoomed in. But it won't be corrupted or removed Just unzoom to have it.

albertolg, 2017-01-29:
downloaded and tested, working good... can you please add also for the X Y and Z ? :-)

sthalik, 2017-01-29:
Is there a point at all to add for position?

albertolg, 2017-01-29:
well... right now mine is this.. I don't know if is useful or not adding it 
[image]

other are working good and easy to edit

[image]

petsild, 2017-01-29:
Hi,
when I see a new change to work with curves :) I would like to ask if it was possible to create a switch like has graphically now the X-axis do to the Z axis doesn't start at the center of 50% but ranging from 0% - 100% explain why.
In all simulators DCS,ROF,BOS.BOM I'm trying to set the view with the least zoom, and shift in the cockpit for a better overview of what's happening. The wider shot is just clearer and there's more to see.
It would be an improvement for a lot of pilots, thank you for any response. Pet

sthalik, 2017-01-30:
@petsild I understand. But we're scaling movement along the X axis on the mapping and not along the Y axis. Maybe I can reduce snap when zoomed in. Stay tuned. I also found some tiny tiny bugs in the initial experimental build, so fixing.

petsild, 2017-02-03:
I thought to make it a manual shift seats in the cockpit using some gate valves. You can now simulate at startup centering to zoom in and then zoom out to the starting position. But when the new centering is the default position back to 50%.

The idea in graphics:
[image]

sthalik, 2017-02-03:
@petsild the zooming exists so using only the initial part of the mapping is easier and more precise.

I'll add zooming XYZ.

petsild, 2017-02-03:
I'm sorry I expressed myself badly only shift the Z-axis in the cockpit.

sthalik, 2017-02-03:
@petsild can you rephrase your original message please?

petsild, 2017-02-03:
Idea to shift the centering of the field of view to be sliding freely forward or backward as the seat of the pilot.

sthalik, 2017-02-03:
But how can you shift it? It's an absolute position taken from raw data.

petsild, 2017-02-03:
I'm not enough versed in the issue of raw data it will probably be a complication to write the correct interpretation.

petsild, 2017-02-03:
If it was possible to impose after the centring of the specific games a different position of the Z axis?

sthalik, 2017-02-04:
@petsild camera -> custom center pose

petsild, 2017-02-04:
Such an option in the settings of glossed over and didn't know what exactly is intended. Thank you very much and will attend to this.
 Have a nice weekend. Pet

petsild, 2017-02-04:
Shift using a custom center pose is a very useful feature. Activation happens to the Y-axis is too sensitive and dependent on the axis of the pitch instead of just turning around the center. You can compensate for the blunt curve of it is not so accurate.
[image]

sthalik, 2017-02-05:
See today's build for zooming mappings implementation. <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>>.

@petsild the custom center pose isn't mapped, and mappings don't take it into account.

albertolg, 2017-02-05:
@sthalik changing the scale do nothing to the zoomings X Y and Z

sthalik, 2017-02-18:
@albertolg fixed in time for 2.3.1. Please report any issues.

albertolg, 2017-02-20:
seems all ok, great work @sthalik thanks again

closed 2017-02-20 by albertolg