opentrack/opentrack #536: Need information about input OT data

pszyrek, 2017-01-25:
Siema, widziałem że można sie po polsku porozmawiać, wiec próbuje :D  Otóż pracuje nad własnym headtrackingiem opartym o czujniki akcelerometru, żyroskopu(narazie bez magnetometru). Zebrałem potrzebne dane, przekonwertowałem je na takie jakie pokazuje debug log w freepie IMU na androida.(zdjęcie w załączniku) Chciałbym wiedzieć czy takie dane wystarczą i są gotowe do wysłania na opentracka, jest ktoś w temacie i dałby rade pomóc?

[image]

sthalik, 2017-01-28:
You can send this data alright via the UDP protocol.

https://github.com/opentrack/opentrack/issues/13#issuecomment-30169073

and my next message in the same thread.

You need to use `struct.pack` rather than `unpack` when sending however. See Python's `struct` module API.

Your accelerometer readings aren't suitable for sending to opentrack as they are. Please substitute all zeros for x, y, z.

sthalik, 2017-01-28:
Just before I forget, send the data in degrees and not radians. It's a legacy flaw of our UDP protocol.

Just send six doubles over the wire hopefully not having to worry about buffering in that stream.

pszyrek, 2017-01-31:
That's my raw data values from sensor:

x, y, z - Accelerometer
yaw, pitch, roll - gyroscope
rotx, roty - rotation from accelerometer

x: -1003.3890 y: 14.7010 z: -60.8170 yaw: 3.89 pitch: -1.12 roll: -1.23 rotx: 0.838 roty: 86.432
x: -1005.6460 y: 22.8750 z: -60.8170 yaw: 3.89 pitch: -1.12 roll: -1.23 rotx: 1.301 roty: 86.303
x: -1006.9880 y: 15.6770 z: -61.1830 yaw: 3.29 pitch: -4.03 roll: -3.54 rotx: 0.890 roty: 86.411
x: -1006.9880 y: 15.6770 z: -55.6320 yaw: 3.36 pitch: -5.88 roll: -6.76 rotx: 0.891 roty: 86.715
x: -1005.2800 y: 10.1260 z: -55.6320 yaw: 3.36 pitch: -5.88 roll: -6.76 rotx: 0.576 roty: 86.781
x: -1009.3060 y: 8.3570 z: -56.9130 yaw: 3.85 pitch: -5.18 roll: -6.90 rotx: 0.474 roty: 86.738
x: -1009.3060 y: 8.3570 z: -63.6840 yaw: 4.27 pitch: -3.50 roll: -7.04 rotx: 0.473 roty: 86.359
x: -1003.6330 y: 12.0170 z: -63.6840 yaw: 4.27 pitch: -3.15 roll: -7.04 rotx: 0.685 roty: 86.305

I have this value, which one i will need?

sthalik, 2017-02-01:
Look at AHRS algorithm for sensor fusion between gyro and acc:

https://github.com/AndersMalmgren/FreePIE/blob/095d48119abcb77fac358aab63a13a92c21d3737/FreePIE.Core.Plugins/SensorFusion/MahonyAHRS.cs#L229

pszyrek, 2017-02-01:
Okay, can i use kalman filter rather than mahony?

sthalik, 2017-02-01:
Technically yes, but you won't gain anything by that and only add complexity.

In the end pass zeros for position and Tait-Bryan angles (0->+-180) as ypr, in that order. Our rotation order is ZYX by the way, see `compat/euler.cpp` for transforms you may want to reuse when converting to Euler.

pszyrek, 2017-02-06:
Hello. I buy another sensor named minimu-9 v3 and i find library for that:
https://github.com/DavidEGrayson/minimu9-ahrs/wiki

It looks pretty good, because libraryallow us to show ahrs data, and show converted data to angles in degrees(pitch, yaw and roll). Library have an option which allow us to vizualize how it works, like this:
https://www.youtube.com/watch?v=8daR-wtQIx0

If i get data from sensor as euler(angles in degrees) Will i be allowed to send this thata by udp to opentrack and that will work?

sthalik, 2017-02-10:
> If i get data from sensor as euler(angles in degrees) Will i be allowed to send this thata by udp to opentrack and that will work?

Yes. Really. Totally.

closed 2017-12-02 by sthalik