opentrack/opentrack #537: check permissions for your .ini directory   ??

t1rta, 2017-01-26:
I used to run this program with no error.
However when I try to run it now I get this "check permissions for your .ini directory" error instead.

[image]

how do I fix it?
please help.

ps I have tried to uninstall and reinstall opentrack but the problem remains.

sthalik, 2017-01-28:
opentrack asks Windows for the user's "My Documents" path. Does it exist and is Windows aware of it?

t1rta, 2017-01-28:
thanks for the reply.
can you tell me what is the exact address for the directory?
and is it possible to move this directory somewhere else?

Emton, 2017-01-28:
C:\Users\YOURUSERNAME\Documents\opentrack-2.3

sthalik, 2017-01-29:
Windows holds the path in registry so you can specify an alternate path. But this is heavily outside the scope of the opentrack project. I won't provide support wrt broken user profiles.

Other than that, a "Documents" directory is a standard Windows feature and we won't jump through hoops if it's missing.

t1rta, 2017-02-02:
thanks sthalik and Emton for all the replies.

yes I have found the culprit.
that directory is in external hdd, when I forget to connect it, then I
get the error.

however, I am still confused how to move it to my internal hdd,
but as you said that this is a window issue not yours,
so I guess I have to google it.

thanks again for all the help.
I really appreciate it very much.

sthalik, 2017-02-02:
@t1rta see what kind of object "My documents" is on the primary hard drive. Try recreating the directory with the external disk unplugged.

t1rta, 2017-02-03:
I am sorry I do not really understand "what kind of object".
can you explain it a bit more?

sthalik, 2017-02-27:
@t1rta delete the shortcut to the external drive. Having such basic directory missing will break many programs.

closed 2017-02-27 by sthalik

fuzzy17, 2017-04-02:
I created c:\users\myuser\Documents\opentrack-2.3 and now get "check permissions for your .ini directory "c:/users/myusers/documents/operntrack-2.3." exiting now".  I treid running both as admin and non-admin and got same error both times.

sthalik, 2017-04-03:
Are you able to access `My Documents` as Windows recognizes it through File Explorer?

fuzzy17, 2017-04-08:
No I did not have.  So I created  c:\users\myusers\My Documents and still get same error.  I checked permission and it shows full control for my user that I am logged in as. Yet I am in admin mode.

sthalik, 2017-04-08:
opentrack doesn't look up for a hardcoded "Documents" name or anything of the sort. opentrack asks Windows to designate the "Documents" directory for the current user. That you had to create it is by itself a red flag.