opentrack/opentrack #538: ATS no moving action

ogimanas, 2017-01-30:
Hi all,
 I have problem in American Truck Simulator x64 and x32. The head movement does not translate to game (mouse and other controls working good, not freezing). Opentrack works fine octopus moving as should. I`m using selfmade 3point hat with IR diodes and Logitech webcam C170 (IR filter removed). No errors no notifications about something is wrong. Tryed ETS x64 and x32, but everything is working very good, no complaines here. So I was trying to search for similarities in games configurations files and control .ini files,  but nothing could find to help solve the issue. I`m desperate, Please, help ideas anything that could help.

MathijsG, 2017-01-30:
You already start the tracking before starting the game? No problems in ETS2, which is technically the exact same game.

ogimanas, 2017-01-30:
OT starting before and after no changes. I know they are the same and thats confusing me most. I`ll try replacing ATS  config and (or) control .ini file from ETS.

sthalik, 2017-02-21:
Ensure freetrack protocol selected.

@MathijsG if additional game settings necessary, can you specify them please?

MathijsG, 2017-02-21:
American Truck Simulator (ATS) and Euro Truck Simulator (ETS) should work quite out of the box.

sthalik, 2017-02-22:
@MathijsG  Thanks.
@ogimanas can you confirm?

ogimanas, 2017-02-22:
The funny thing happened. For some time I played ETS2 (the working OP) by mistake launched ATS and its working perfectly. Dont know what  happened, but OP working good and I enjoying playing ATS. Thanks guys for support, Super community. 
@sthalik Im always using FT protocol.

ogimanas, 2017-02-22:
I think the issue is solved, but I leave this thread for a week and than will close. Thank again :)

sthalik, 2017-02-23:
That's ok. Glad it's working for you.

closed 2017-03-02 by ogimanas