opentrack/opentrack #539: REQ. Offset dead zone

WF000, 2017-01-31:
Hi, Halik.

When you able, can you consider this request, offset dead-zone.
Like trimmed custom size dead-zone with no interpolation. Like in TrackIR software.
Thanks

[url=https://postimg.org/image/n8z9akuf1/][img]https://s30.postimg.org/n8z9akuf1/Clipboard_2.jpg[/img][/url]

sthalik, 2017-02-01:
@WF000 the Catmull-Rom splines won't let you do it easily, not without serious workarounds.

Also I'm opposed to deadzones, even at the start of the mapping. Consider a sigmoid (logistical function) mapping rather than a pure deadzone.

WF000, 2017-02-03:
I hear you... yes workarounds... came to mind to put lots of dots where dead-zone would be on mapping, to simulate straight line...will try. & ok , sigmoid will put to test..  
Need 'perfect' still look on one point , sigmoid opposed even if small jitter... more filter latency tweak too I guess

sthalik, 2017-04-06:
@WF000 Now Accela is three-axis rather than 3 times one-axis. Can you confirm it works any better now?

WF000, 2017-06-08:
Hey, 
sorry i took my time, have been busy (no time for flying) ..
then, out of nowhere, computer power supply eXploded 3 times,
condenser and both mosfets cracked ..lol
yeah, lol, another 100bucks , and luckily no serious damage...

anyway back to stuff, .. much better, Thanks, (ver 2.3.7) now can do much better
interpolation, almost straight line, if dots are longer apart.. but it suffices just nice
Thanks again, cheers Halik
ps. may close issue if want

closed 2017-12-02 by sthalik