opentrack/opentrack #540: Bad return to center

jcrtalent, 2017-02-01:
On latest opentrack with settings:

Tracker: joystick, all axis disabled except pitch and yaw, mapped to analog mini joystick of the thrustmaster FCS throttle.
Protocol: Freetrack (hide freetrack, show TrackIR)
Filter: Accela default settings (but the problem also happens with no filter)
Default linear axis mapping.

The problem is that basically after the first input just after tracking started, on joystick release and raw data showing all 0, game data is not at 0. This happens always but sometimes the offset from 0 is not as important as to invalidate the session. Other times I have to stop & start tracking again. The offset seems to stick throughout the session, meaning that if whatever offset I get after first input is more less what I am going to have throughout the session.

Being a joystick input tracking and having raw data 0, I expect a game output data of 0. What am I missing?

sthalik, 2017-02-01:
Make sure "center on startup" disabled then confirm still broken.

sthalik, 2017-04-06:
I'd rather filter before centering but that causes its own share of problems.

closed 2018-05-14 by sthalik