opentrack/opentrack #541: VS2015 "The application was unable to start correctly(0xc000007b)"

yyyolanda, 2017-02-03:
I tried to build Opentrack in my Win10. I used MVS2015 Community and the latest source Opentrack 2.3. After building opentrack through CMake , I tried to run it on VS2015 and it said "The application was unable to start correctly(0xc000007b)" and " The program '[5552] opentrack.exe' has exited with code -1073741701 (0xc000007b)."

When I was searching solution online and I got "The 32 bit application cannot load a 64 bit dll". But I followed the guiding of Opentrack Building...
Is it because my computer is 64bit but the source code is for 32-bit?

yyyolanda, 2017-02-03:
On Output it also said 'opentrack.exe' (Win32): Loaded 'C:\Windows\System32\opentrack-migration.dll'. Cannot find or open the PDB file.

sthalik, 2017-02-04:
@yyyolanda run the `INSTALL` target.

yyyolanda, 2017-02-05:
@sthalik You mean run the INSTALL target only? I have built INSTALL target, and I set the opentrack target as the startup project. When I try to run the whole project, it said "The application was unable to start correctly(0xc000007b)".

If I set other target as the startup project, it said"it is unable to start program" and I  thought it is because there is no executable file inside.

sthalik, 2017-02-05:
Run the executable that installed as part of the INSTALL target.

yuanchang200, 2017-02-06:
@sthalik I have built ALL_BUILD in both debug and release, there are no errors. Then I build the INSTALL target, it has error MSB3073:

  file INSTALL cannot find "C:/opentrack/opentrackvs2015/nl_NL.qm".
24>
24>
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: The command "setlocal
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: "C:\Program Files\CMake\bin\cmake.exe" -DBUILD_TYPE=Debug -P cmake_install.cmake
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorlevel% neq 0 goto :cmEnd
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmEnd
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmErrorLevel
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: exit /b %1
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :cmDone
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: if %errorlevel% neq 0 goto :VCEnd
24>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5): error MSB3073: :VCEnd" exited with code 1.

I google it, according to [here](http://stackoverflow.com/questions/31457306/visual-studio-msb3073-error-exited-with-code-1), I change the project to new directory and run the Visual Studio 2015 in Admin Mode, but none of them worked. What's wrong with it? Thank you for helping me.

yyyolanda, 2017-02-07:
Thanks! I solved the problem!

closed 2017-02-07 by yyyolanda