opentrack/opentrack #542: Opentrack work but the camera doesn't move ingames (octopus is moving)

ppe201, 2017-02-06:
Hi , so i just bought the app Trinus VR (wich work well for most games) but it happened that the app doesn't work in Dirt Rally and Project Cars , so i tried using the "TrackIR" option (wich is supposed to work on dirt rally but doesn't) and also tried everything on project car (wich is supposed to work fine with OpenTrack but doesn't either) so i'm a bit lost.

I did every settings , i linked my phone to it the octopus is able to move in the app so the position of the phone is tracked but it's not showing ingame (camera doesn't move a bit)

ppe201, 2017-02-14:
anyone ?

Forsaken-Player, 2017-02-16:
Did you try running in Admin mode?

sthalik, 2017-02-21:
Admin privileges aren't necessary for that.

See for instance: <<https://steamcommunity.com/app/234630/discussions/0/613957600526874697/>>

closed 2018-01-04 by sthalik