opentrack/opentrack #544: Can't find RealSense SDK 2016 R2 installer, not working with SDK 2016 R3

SnorreSelmer, 2017-02-09:
Been trying to see if I can get OpenTrack to use my Razer Stargazer (RealSense R300) for face-tracking, but it doesn't work. Unsure which component is missing, have installed the R3 SDK face-track component. Screenshot attached.
[image]

JayMonkey, 2017-02-10:
it does work but needs a but of nursing with the drivers ...

You must be running Windows 10 x 64 

First let windows install the base (usb 2 & 3 hub) drivers ... should auto detect .. next install latest RS300 Real Sense Camera Depth Manager from here ..

https://downloadmirror.intel.com/25044/eng/intel_rs_dcm_sr300_3.3.27.5718.exe

Finally download real sense 2016 sdk runtime R2 from here ...

http://registrationcenter-download.intel.com/akdlm/irc_nas/9078/intel_rs_sdk_offline_package_r_10.0.26.0396.exe

run install and select base components + head tracking plugin ...

Things have changed quite a bit with the RS300 (Stargazer) cameras and R3 & SDK software  since the initial opentrack support three years ago ... it can be a bit flakey but when it works it works well.

Am going to post request for next build of OT to support/use latest R3 drivers and SDK

Cheers
Jay

sthalik, 2017-02-12:
I'll make an unsupported build for the time being, along with instructions on downloading the SDK parts for RealSense. But it can take a while given I'm out of action.

JayMonkey, 2017-02-17:
Hi sthalik ...

That would very much be appreciated ...  i understand what you mean with regards to the drivers and runtime but it's pretty much the same for anything using realsense rs200 or rs300 series camera chipsets .. even the new Stargazer requires downloading and installing intel SDK runtime so i would not worry about that too much.

Anyone with a real sense camera should be used to the process .. 

Look forward to your new build ... and am happy to beta test once your ready for some feedback.

Cheers
Jay

sthalik, 2017-02-19:
In the test builds directory <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>>

The build is `opentrack-test-20170219_1-realsense-r3-tmp-do-not-use`. I won't be releasing the code because @ph0b owns the module's maintainership.

sthalik, 2017-02-19:
Keep in mind that it'll still ask for downloading the old SDK if R3 is missing. I only moved it to R3, but the rest of the code stood intact.

JayMonkey, 2017-02-19:
@sthalik ,

Great and fast work  dude .. will have a look-see in the coming week and get back with some feedback

SnorreSelmer, 2017-02-19:
Excellent! I tested it, and while the settings-panel for the RealSense plugin didn't work, tracking worked great! I also tested it with DCS 1.5.6.1938, and it works there too (had to invert the yaw and roll axies, but all else was fine)! Now I just need to tweak the profile.

sthalik, 2017-02-21:
What is the settings panel? Not having the device I can't check. Thanks.

SnorreSelmer, 2017-02-21:
I have no idea what's supposed to be there, it's the one you get when
clicking on the hammer next to "RealSense" in OpenTrack. It just says
"Install RealSense SDK R2"

sthalik, 2017-02-21:
Like it says "no configuration available at the moment". There are no user-visible options at all.

As for yaw and roll inversion, we'd need to investigate if it's a change in the SDK, model-specific, or DCS-specific. Please remember not to use any headtracker lua changes. Also use the freetrack protocol just like that.

sthalik, 2017-02-21:
You may also see if it also needs inverting in some other game. For instance Falcon BMS is free.

JayMonkey, 2017-02-21:
There are no tracker settings (in opentrack) when using realsense device ... as far as i know

You just need to make sure you have the realsense depth manager installed (r3) and the sdk runtime for head tracking (r3) download both at the intel realsense support site

SnorreSelmer, 2017-02-21:
The tracking works fine, just the configuration (if there is one) won't
come up until I install R2. I have installed the components Jay mentioned.

The misbehaving axes are misbehaving in OpenTrack. I had the same issues in
Elite Dangerous, so I just inverted the axes in OpenTrack to solve it once
and for all. ;)

JayMonkey, 2017-02-26:
Just a heads up for everyone coming from real sense r2 to 3r run times ....

It looks like there is a lot of code optimizations in the r3 runtimes with regards to head tracking .. seems intel are making use of some cpu features that are only current in later gen CPU's.

So if your real sense headtracking worked for you before with the r2 runtimes and the release version of OT but it doesn't work now chances are that your cpu can not run the r3 optimizations in which case you will have to stick with the r2 run times and current release of OT.

An easy way to test is to run the intel R3 sdk samples browser included in the r3 SDK runtimes .. if it runs all the other demos such a raw stream and camera ir demos but fails to run the head tracking demos then chances are its a cpu gen thing ...

I believe Intel recommend SKL gen cpu and chipset ... for rs300 camera and r3 sdk .

My old dev system was a sandy bridge/Z77 system and ran the old version of OT with real sense r2 sdk fine but it will not run r3 head tracking ... moved to newer hardware system and seems to work fine ...

If anyone else can confirm the above then that would be cool for others but thats what i'm seeing so far with this dev build of OpenTrack and RS300/StarGazer camera with real sense SDK R3. 

On a positive note the r3 tracking does seem much better and faster with less over head on the cpu. 

Cheers
Jay

SnorreSelmer, 2017-02-26:
Yeah, Razer's system requirements for the Stargazer is a 6th gen. Intel CPU
(Skylake).

sthalik, 2017-02-26:
Are you guys getting a crash on R3 with no device plugged in?

SnorreSelmer, 2017-02-26:
I get this error when trying to start tracking without the Stargazer connected: 
[image]

JayMonkey, 2017-02-26:
I don't think the real sense drivers and framework will be loaded by windows if no device connected ... ???

sthalik, 2017-02-26:
It crashed for me in some nonsensical way. I'll investigate further.

JayMonkey, 2017-02-26:
>  SnorreSelmer commented 2 hours ago
> Yeah, Razer's system requirements for the Stargazer is a 6th gen. Intel CPU
> (Skylake).

Yeha .. I read that too but it makes you wonder if it really is necessary ??
That why i asked for confirmation form others running on < 6th gen systems

Please anyone testing on less than 6th gen intel CPU please give feedback 

Cheers
Jay

sthalik, 2017-02-27:
They might actually depends on some newer SSE/AVX variants that work only on the last-gen Intels. But of course this is unnecessary, given that they could provide an executable without these optimizations working on all crappy hardware including the first Pentium 4 model.

JayMonkey, 2017-02-27:
Have a few different systems i can test it on ... will try and figure out what works .... and what doesn't

Jay

ph0b, 2017-03-30:
There are many drawbacks for us going to R3: losing Sandy Bridge support and not having the small online installer for SDK runtime (users will have to install the complete SDK and relying on SDK install will introduce incompatibilities as soon as a new one is released or other apps do the same).

@JayMonkey can you test again the regular build of opentrack with R2 SDK runtime core+face modules installed from here: https://github.com/opentrack/opentrack/blob/unstable/tracker-rs/redist/intel_rs_sdk_runtime_websetup_10.0.26.0396.exe?raw=true on your recent system?

I'd like to know if it works with your StarGazer and if the performance is really different between R2 and R3 with Skylake+.

If it works and perf is the same than with R3 on a recent system, I'll choose to hold on to R2 as it's much better for regular users. (btw the R2 runtime is normally installed by opentrack installer and on a click on "install runtime" on tracker error and config windows, but it's broken in current release - fixed in code)

sir-maniac, 2018-06-03:
In case anyone is interested, I found documentation in the 2016 R2 developer guide (/doc/PDF/sdkdevguide.pdf in the SDK) that gives instructions on how to make a custom SDK runtime install.

[I put one together with only the core and face components](https://drive.google.com/open?id=1kJ3o9vFbKmHRa21p6tBTWOr5wk91eUjx).  It uses the official install, compressed down into a self-extractor, and still requires acceptance of the EULA, so it appears to comply with Intel's permission to redistribute.  It's 279MB, much smaller than the 1.8GB offline SDK.  So a file that can possibly be offered as an opentrack download.

Also, there might be a way to mirror the individual componenets, and offer a runtime install that downloads them, giving a nice compact websetup(2MB). This might be possible, since the download location appears to be configurable in the setup.

sir-maniac, 2018-06-03:
There might be a way to offer support for both 2016 R2 and 2016 R3 at the same time: 

The code that acquires access to the runtime appears to be located in the liberally-licensed libpxc source code(3-clause BSD-like).  It could be modified so a configuration option can be selected to choose which framework to used.  Class inheritance can encapsulate the differences in each API.

I'd be glad to look into implementing this functionality if there is enough of an interest.

sir-maniac, 2018-06-17:
Hi @sthalik, @ph0b:

I'll just leave the steps to produce the offline installer here 

When run, it extracts to the %TEMP% folder and automatically runs the official setup.exe, and displays the intel EULA, just like an official installer.

<details><summary><b>[expand me]</b> </summary><p>

1. Per the documentation under "Installing SDK runtimes" subheading "Installation Case 2: Module Offline Installation", run the --pre-bundle command line option to the 
    ``` > intel_rs_sdk_runtime_websetup_10.0.26.0396.exe --pre-bundle E:\runtime-setup ```
2. In the installation dialog, accept the EULA, select "Core" and "Face Tracking" only, click Finish.
3. extract the full SDK offline installer to a folder with the `--extract-to` option
4. copy over the following files from the extracted folder (\installs) into E:\runtime-setup\installs, all the files specified are required for a successful offline installation.
    - `intel_rs_sdk_runtime_*_core_*.msi`
    - `intel_rs_sdk_runtime_*_face3d_*.msi`
    - `intel_rs_sdk_runtime_*_lanternrock_*.msi`
    - `vcredist_*_v110.exe`
5. Pack the installation files into a self-extracting exe.  I chose 7-zip for size and simplicity.  Step to do that are:
    1. Compress files with 7-zip, I used the "Maximum" compression level.
    2. Source the file 7zS2.sfx from the [latest 7z LZMA API](https://sourceforge.net/projects/sevenzip/files/LZMA%20SDK/lzma1805.7z/download)
    3. make a config.txt file with the following contents (the command line options come from the "readme_cmd.txt" from the `--pre-bundle` created folder):
        ```
        ;!@Install@!UTF-8!
        Title="Intel RealSense Face & Core SDK Runtime 2016 R2"
        BeginPrompt="Do you want to install Intel RealSense Face & Core SDK Runtime 2016 R2?"
        RunProgram="setup.exe --fnone=all --finstall=vs2012rt,lanternrock,core,dummy_db,face3d"
        ;!@InstallEnd@!
        ```
    4. Run the following command to make the executable:
    ```>copy /b 7zS2.sfx + config.txt + archive.7z archive.exe ```
</p></details>

sthalik, 2018-06-18:
While it's legal to redistribute the web installer, it's illegal to redistribute the offline installer. We *could* run the commands you specified to install the files on the user's computer. 

@ph0b any reasons not to? I can do it if it's ok.

ph0b, 2018-06-18:
@sir-maniac great job in bringing down the size of the offline runtime installer! I wasn't expecting such gain in size with these files.

I confirm it's permitted to redistribute it as long as the EULA piece is accepted by the user. It's the _SDK_ which can't be redistributed, not the _SDK runtime_.

Regarding the modules to include, only core and face3d are required. I'm not sure regarding the VS redistributables but there is at least no need to include the one named lanternrock.

sir-maniac, 2018-06-19:
> Regarding the modules to include, only core and face3d are required. I'm not sure regarding the VS redistributables but there is at least no need to include the one named lanternrock.

@ph0b the VS redistributables and lanternrock are both used when only "core" and "face3d" are selected in the install.  If they are not included in the package, the installer attempts to downloads those packages and install them anyway.

I think the only way around this would be to hack the setup configuration files, or make our own installation package.

hsinyu-chen, 2019-11-10:
any update on this? 
can't get R2 installed on win10/2700x

claudio-pires, 2020-02-06:
Hi all, does anybody have a copy of Intel RealSense SDK 2016 R3 installer?
Thanks in advance!!
Claudio

sthalik, 2020-03-29:
I can't find the installer either. It's behind paywall on Intel's site and not available elsewhere. There's no point in upgrading to R3 (which is actually easy) unless users can get the SDK.

Luk164, 2021-08-17:
Anyone ever manage to find one? The modules are EOL so they are now dirt-cheap but we cant use them w/o SDK...

sthalik, 2022-10-21:
See if this works for R2: https://www.dropbox.com/s/bb8cpv5n9b4ptpu/opentrack-test-20221021_01.7z