opentrack/opentrack #545: Request for next OT build to support latest Real Sense Drivers and SDK

JayMonkey, 2017-02-10:
As stated in last issue OT does not work with latest Real Real Sense SDK runtime R3 ...

With the availability of consumer real sense cameras such as the Razer StarGazer it would be great if OT supported the latest drivers and SDK runtimes ... it can be persuaded to work but its a bit of a mix and match and causes OT to crash occasionally.

Right now you have to use latest R3 Depth Manager with 2016 R2 SDL runtimes ... R3 runtimes are a lot more efficient and make use of the hardware improvements in the RS300 chipset.

OT + RS300 based 3D camera gives ultimate PC head tracking solution in my opinion, no wires or batteries, no bits of paper or phones strapped to head ... full 3D rotational & positional  (6 axis) high speed tracking.

I've tried OT with all solutions including some custom hardware that i've made, but real sense cameras are the future now that the tech is stable and generally available to consumers ...

Happy to beta test a build and give feed back if needed ...

Cheers
Jay

paulnd, 2017-02-11:
+1 please

SnorreSelmer, 2017-02-11:
+1, and will beta-test too!

sthalik, 2017-02-12:
Hey,

Testing isn't as much of a problem. The issue here is that there are no redistributable files provided for the end-users, yet. You'd have to get the full SDK on your own and install relevant plugins.

sthalik, 2017-02-12:
Duplicate of #544

closed 2017-02-12 by sthalik