opentrack/opentrack #546: Impossible to use with TIR3Pro?

Phil8811, 2017-02-16:
Hi, i can't get to make TIR3 working, it seems it can't put TIR3 "ON".
When i use Natural Point application it works but not with Opentrack, it's not detected and led stay "OFF" with Opentrack
I would like to use Opentrack and not the Natural Point software for use it with RiftCat (Opentrack+Phone) for get x,y,z data from TIR and rotation from Magnetometer, do you know how to do please?

Best Regards

PS: i didn't found a solution working for me on Google

sthalik, 2017-02-16:
1) The software doesn't support TrackIR devices as input
2) We'd like so too, but the XYZ input becomes incorrect after just a few seconds so much that it's useless.

closed 2017-02-16 by sthalik

Phil8811, 2017-02-16:
Hi thx for fast answer
I understood your answer but maybe it could help you to fix it:
It seems some people got TIR3 with Freetrack working:
https://forum.il2sturmovik.com/topic/12829-great-news-freetrack-users-here-how-it-works-now/

Best regards

sthalik, 2017-02-16:
Use old_trackir = 1 for freetrack, zero for opentrack and ftnoir.

Phil8811, 2017-02-16:
Ok don't know why but for me Opentrack didn't open the TIR3 it stays light off
Thx for your help, I imagine i must find another solution

Have a great day

sthalik, 2017-02-16:
opentrack doesn't have any support for TrackIR devices. It won't turn itself on with a program, such as opentrack, not supporting the device.