opentrack/opentrack #547: Send data from rasppi python script to opentrack via udp

pszyrek, 2017-02-16:
Hello. I have data in euler, and now i want send this by udp protocol to opentrack but i don't know how could i send printed data via udp. I have raspberry pi 2B and my sensor is minimu-9 v3. Data from sensor in attach file.
[image]

Could you help me?

sthalik, 2017-02-16:
Send six double-precision floating-point values, each 8 bytes total. In order, x, y, z, yaw, pitch, roll. Are we good?

pszyrek, 2017-02-16:
No dobra, ale nie bardzo orientuje się jak napisać skrypt który wyśle mi przez protokół udp to co mi printuje mój skrypt :/

sthalik, 2017-02-16:
Zależy od języka. W pythonie na przykład moduł `struct`.

pszyrek, 2017-02-16:
Zajebiscie, o to mi chodziło :D Nie masz w zanadżu czegoś co przybliżyłoby mi jak napisać skrypt który to wyśle? Jakis tutorial albo cos w tym stylu?

sthalik, 2017-02-16:
Gdzieś w zamkniętych issues coś jest ale poza tym to nie. Ale w google znajdziesz w minutę.

pszyrek, 2017-02-16:
Dodałeś taki skrypt, powiedz mi to "foo" to nazwa pliku który printuje to co chcesz wysłac?

`#!/usr/bin/env python

import io
import struct

with io.FileIO("foo") as stream:
    while True:
        buf = stream.read(6 * 8)
        if len(buf) != 6 * 8:
            break
        print struct.unpack('dddddd', buf)

(12.808689804246633, -10.59810890650144, 6.944386467079594, 5.035714626312256, 57.694427490234375, 12.76200008392334)
(12.808689804246633, -10.59810890650144, 6.944386467079594, 5.035714626312256, 57.694427490234375, 12.76200008392334)
(11.755536289623992, -10.651814457389555, 8.071451630922075, 1.735714316368103, 54.74729537963867, 12.760000228881836)
(12.291474781644514, -10.638413448470189, 7.507529822354849, 3.385714530944824, 56.22663879394531, 12.76200008392334)`

sthalik, 2017-02-17:
Ja tutaj piszę do pliku "foo". Ty masz pisać do socketa UDP cały datagram o takiej postaci jak `struct.unpack` wysrywa.

sthalik, 2017-02-17:
Ja tu robię `struct.unpack` czyli czytam z pliku. ty masz zrobić `struct.pack` i ci zwróci zawartość datagramu co masz wysłać. Oczywiście musisz wiedzieć jak działa `struct.pack` i obsłużyć socket UDP.

pszyrek, 2017-02-21:
Okej, musze w jakiś sposób modyfikować skrypt który mi printuje dane? czy wystarczy ze uzyje struct do przesłania wyprintowanych danych do opentracka?

sthalik, 2017-02-21:
Nie.

Taka pomoc jest poza zakresem issue trackera.

closed 2017-02-21 by sthalik