opentrack/opentrack #552: Slow start of tracking

MathijsG, 2017-02-21:
Although I'm happy to see this new release, I discovered a significant slow down of the initial start of the tracking. When clicking on start the CPU load rises very high and after about 5 - 6 seconds I get working tracking and seeing three crosses.

Is this expected behavior in the new release? Because of the weird flippings I get sometimes with hat tracking (still get them in this release unfortunately, https://github.com/opentrack/opentrack/issues/289) I rely on the restarting of headtracking instead of just resetting the view with centering (because then the weird flip stays). However with restarting the tracking taking up 6 seconds now this is not really an option anymore.

sthalik, 2017-02-21:
See if it goes away if you unplug and plug in the camera again. The tracker tries to grab frames like a rabbit on a mission in case capture fails the first time. This in itself is suboptimal but let's make sure what's the issue first.

MathijsG, 2017-02-21:
I replugged and the issue is still present. Reverted back to 2.3.0 (by just installing the older version over the newer) but now the issue is also present in 2.3.0. Will try with removal tomorrow.

sthalik, 2017-02-22:
Did you plug any more devices into the usb controller? Even if not, try shutting down the system, flicking the PSU switch or pulling the plug after it's closed, wait for capacitors to empty, and only then reboot.

If this is a software problem, switch the USB port and it'll force drivers to reinitialize.

sthalik, 2017-02-22:
@MathijsG can you check if 2.3.1 hotfix candidate broke any of the following?

- PT preview window, possibly crashes
- PT accuracy due to meanshift coefficient change
- video preview CPU usage
- possible crashes when opening module (filter/input/output) dialogs

The hotfix is meant to fix changes not reverting after pressing "cancel" in modules. But the other changes are fun and relatively benign so I'd like to put them into it as well.

See <<https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa>> for today's build. Goddamn I wish it had a short address!

sthalik, 2017-02-22:
It's gonna have a short URL after DNS refreshes. I really love this free DNS service!  ❤️

sthalik, 2017-02-22:
And now it does. http://opentrack.misaki.pl

sthalik, 2017-02-22:
Note that a workaround for your issue's in the build also.

MathijsG, 2017-02-22:
Somehow it's still present and it also affects the other programs like FaceTrackNoIR, so I think something is messed up at my side. Will investigate further.

MathijsG, 2017-02-22:
Already found the problem. The Geforce Experience spyware from Nvidia is delaying the webcam significantly when using the sharing functionality of this software to be able to easily record things.

sthalik, 2017-02-23:
@MathijsG the workaround I added is sound, given that some of crappy cameras refuse to submit the few initial frames. That removes busy-looping from at least these types.

In the future, the "sysinternals procexp" can help you see a module list for a running process. Crapola such as "Gigabyte OC GURU" will manifest as additional .dll in the module list.

Thanks for letting me know of the real cause.

closed 2017-02-23 by sthalik