opentrack/opentrack #554: Auto-centre not giving full 180 degrees in both directions

akamustang, 2017-02-25:
You only get full tracking left (0 to -180) and right (0 to 180) if you start the raw axis value at zero, at any other position you lose the full range.

I made a short video demonstrating the issue: https://www.youtube.com/watch?v=iZn3xSJYOxc

sthalik, 2017-02-26:
The reason for the issue occuring is `scaled_rotation` and in particular <<https://github.com/opentrack/opentrack/blob/unstable/logic/tracker.cpp#L235>>.

We do that to get a "canonical" Euler representation in face of the gimbal lock issues. The method actually works except for that singularity you get.

We should do something at <<https://github.com/opentrack/opentrack/blob/unstable/logic/tracker.cpp#L164>> taking into account c_div and c_mult. I'll look into it after the 2.3.1 hotfix.

akamustang, 2017-02-26:
Thanks for responding, I tried to build the source to have a bash at fixing it myself, but wasn't able to get it to compile, that's a topic for another day though.

sthalik, 2017-02-26:
It should work with recent GNU and MSVC 2015+.

akamustang, 2017-05-04:
Just read in the 2.3.1.6 release notes that hopefully this should be fixed.

Not sure what the related commit was, [9958d78](https://github.com/opentrack/opentrack/commit/9958d784532c43e46df19e67c6f4513e332b7acd) maybe?

sthalik, 2017-05-04:
@akamustang apologies, I meant to refer to #600 rather than this issue. There's no rescaling being done in the filter, no.

akamustang, 2017-05-04:
Ah okay, no worries.

sthalik, 2018-05-14:
It should be fine in the development build at <<http://opentrack.misaki.pl>>. It's gonna be part of 2.3.11.

closed 2018-05-14 by sthalik

LeoPazEs, 2018-05-19:
Its working GREAT but Arma 3 Dont Recognize The program you can help me?

mustanggb, 2019-07-01:
@sthalik To let you know I tested in 2.3.11 beta3, and the bug still exists, so this is not fixed yet.

sthalik, 2019-07-01:
@mustanggb can you confirm that you tested on the newest development build from the said address?

mustanggb, 2019-07-03:
Okay you were right, the dev build seems to work fine, my mistake, thought it was in the beta release already. Thanks a lot!