opentrack/opentrack #555: OSX support

tmcsys, 2017-02-27:
I became interested in opentrack because of its potential for use as an assistive technology.  I got here by way of Apple, who feature assistive technology in their products.

I was hoping to find a current binary release for OS X, but finding none, was encouraged by the fact that I was able to build and run OpenCV, and by the statement in the opentrack readme stating that "the codebase builds on ... OS X".  That, however, was not my experience with the 2.3.1 version.  

I was disappointed that I was not able to find any OS X build instructions, either on github or on the net (!), but I was able to glean enough from the MS build instructions to at least get a build working.  So far I haven't run across anything horrible, but there are quite a few nit-picky issues, mostly with things like function declarations, but that are significant enough to halt the build.

Searching through the codebase, I was surprised at how very few conditional sections of code there are to support OS X (now macOS).

Reading through some of the issues here I sense the frustration surrounding OS X support.  It looks like in the past there was a real effort made to support OS X, but based on my codebase scan it looks to me that the idea of supporting OS X is falling off the radar.

If you think it would be helpful, I would be willing to work on resolving some of the OS X build issues and possibly contribute some OS X build instructions.  What I am currently in no position to evaluate is what it may take to try to work through some of the potentially ugly errors that may very well accompany an initial OS X build.

At least if you had a version that could be built on current versions of OS X you'd be one step closer to attracting more developers that could help with debugging and supporting opentrack on Apple products.

sthalik, 2017-02-27:
@tmcsys we'd need someone using OSX daily. I can't possibly support it like I do for Windows or even Linux with a PC and no real need to have it installed anywhere.

sthalik, 2017-02-27:
You're free to edit the wiki.

As for the present state:

- Some keybindings work, other don't. Pressing a keybinding prevents it from getting sent back to the OS.
- Generating the `.app` should still work.
- There shouldn't be build errors.
- No joystick input or keybindings.
- No camera list.

That's pretty sad but it was a lone effort from me not ever needing the functionality. You have it in a pretty decent state to improve. The core work is already done.

sh

tmcsys, 2017-02-27:
Re: There shouldn't be build errors.

Are you able to try a build on 10.11?  I'd like to get some confirmation of what I'm seeing in the 2.3.1 release.

The camera list is about 10 lines of code - selection another 10 or so.  (Apple keeps changing and deprecating their APIs with new versions.) 

Integrating .m code is another issue. It looks like OpenCV doesn't list OS X cameras either.

Yaffa1607, 2017-03-21:
Can you tell me how to install for mac , after downloading?

closed 2018-08-04 by sthalik