opentrack/opentrack #556: Opentrack CSV data logging

JamesXu20150314, 2017-02-27:
Hello,

What's the means of the CSV data logging as follows:
[image]

Also, I want to know the frequency of every row data. (How long time it will spend for a row of data?)

Thanks,

Sincerely,
James.

sthalik, 2017-02-27:
@JamesXu20150314 the data's logged by the tracking pipeline while it performs multiple transformations. It's logged at 250 Hz, and it doesn't matter how often the tracker returns new data.

See <<https://github.com/opentrack/opentrack/blob/unstable/logic/tracker.cpp#L157>> and look for other instancess of `logger`. I hope the code is self-explanatory enough. If you can't understand some transformations in the pipeline, please let me know.

JamesXu20150314, 2017-03-01:
Thank you vey much.
 I just want to use this Opentrack tools, but I will not go over the code.

Sincerely,
James

sthalik, 2017-03-01:
Then you're most interested in the columns `dt`, `raw`, and `mapped`. Also `corrected`->`filtered` shows you filter operation.

lowkeygit, 2017-06-11:
I also have the same question. What's the meaning of corrected, filtered and mapped? 
I have read the link you posted. There is not much comment and I didn't understand the code.

1. "corrected - after various transformations to account for camera position." 
What's the transformations referred here?
Why it needs to be done with these transformations?

2. I found that setting interface also includes "filtered" and "mapped". I choose Accela as filter. However, I didn't know the meaning of these two and also cannot find further information in opentrack's github website.
What's the meaning of "filtered" and "mapped"?

Thanks a lot,

sthalik, 2017-06-11:
@lowkeygit 

> "corrected - after various transformations to account for camera position."

Emitted after centering the view, invert setting applied.

> I found that the data in "filtered" and "mapped" seem the same.

They're only ever the same if you don't modify stuff in the mapping window.

> To my understanding, "filtered" means smoothing the raw data and eliminating the outliers while [...]

Filtering is keeping very similar data almost the same. User doesn't want their screen jumping because they moved for half a degree back and forth each frame.

lowkeygit, 2017-06-11:
1. So, the invert setting will apply on the filtered data set, right?
2. Btw, I cannot decide whether I should set the invert and how to set since I only use it to track the head movement and cannot see any video.

lowkeygit, 2017-06-16:
One more question about the dataset. What's the meaning of "dt"?
I guess that it is a period of time?

For example, 
dt	          	rawYaw	rawPitch    rawRoll
2.199E-06	14.9796	-5.90093  95.0319
0.0107728	15.9796	-6.90093  96.0319

To my guess, it means that:
After passing 0.0107728 **second**, the Yaw has changed from 14.9796 **degree** to 15.9796 **degree**.
Am I correct?

Looking forward to your reply.

Thanks,

sthalik, 2017-06-16:
It's time since tracking start. It's been named `dt` perhaps incorrectly. Keep in mind that the main loop runs 250 times a second so logging generates quite a bit of data.

lowkeygit, 2017-06-20:
If It's time since tracking start, the timestamp should be increasing all the time, right? However, dt has specific numbers, sometimes larger and sometimes very small. Why would that happen?
I'm trying to figure out how it works but still get no clue. Thanks.

sthalik, 2017-06-20:
There should be 250 values each second. The `dt` unit is in seconds. If the tracker can't keep up according to the hires clock, it starts sleeping for less than 4 ms until it still keeps 250 Hz.

lowkeygit, 2017-06-20:
I think I got your point! Thank you so much.

jnhong, 2017-09-12:
Hi,

I also need to know what the data means for a project I'm working on. Could you verify whether what **lowkeygit** said was correct or not, in particular the bolded part below?

> For example,
> dt rawYaw rawPitch rawRoll
> 2.199E-06 14.9796 -5.90093 95.0319
> 0.0107728 15.9796 -6.90093 96.0319
> 
> To my guess, it means that:
> **After passing 0.0107728 second, the Yaw has changed from 14.9796 degree to 15.9796 degree.**
> Am I correct?

If not, could you tell me what the data means?
I just need to know if the Yaw, Pitch, and Roll are in degrees or radians, and if the data is relative to the previous row or is absolute/relative to some origin.

Thanks

JamesXu20150314, 2017-09-12:
Hello jnhong,

Your guess is right. These four column numbers denote respectively time, yaw, pitch, and roll values.

JamesXu

jnhong, 2017-09-12:
Hi @JamesXu20150314,

Unfortunately, that doesn't answer my question.

JamesXu20150314, 2017-09-12:
I am so sorry for you could not understand my answer. The Yaw, Pitch, and Roll all in degree,and which are absolute value.

I think you should do some experiments to solve problem like this.

sthalik, 2017-09-12:
@jnhong the data's dropped after several transformations

- raw, before anytyhing is done
- corrected, after centering
- filtered, after it leaves the filter. same values if no filter selected.
- mapped, after mapping, relative translation and the like

closed 2017-12-02 by sthalik

abhinaynatti, 2018-05-17:
Is there a way to set 'dt' manually?. Thank you.

sthalik, 2018-05-17:
No, the data is always processed at 250 Hz.