opentrack/opentrack #557: settings wont save.

hollywood156, 2017-02-27:
i have been trying to get arma 3 working with open track, for some reason i cannot get the settings to save. please help. btw, i have no programming knowledge.

sthalik, 2017-02-27:
Are you pressing the "OK" button in the settings dialogs?

hollywood156, 2017-02-27:
yes

sthalik, 2017-02-27:
Which settings are you unable to save?

hollywood156, 2017-02-27:
i cannot save anything in the options menu. i tried using portable software and installed software. im using windows 10 as OS.

sthalik, 2017-02-27:
Are you able to create another profile and see it on the profile list?

hollywood156, 2017-02-27:
Yes

sthalik, 2017-02-27:
What's the content of the profile directory? Use the profile dropdown to the left of the profile list to get to the directory.

sthalik, 2017-02-27:
Damn I can reproduce it myself. What a disaster...

sthalik, 2017-02-27:
@hollywood156 released hotfix2. Thank you for a quick report.

closed 2017-02-27 by sthalik