opentrack/opentrack #558: will not load when arma starts

hollywood156, 2017-02-27:
thanx for the fast response with my last issue. after you fixed the last issue, now it tells me "opentrack is taking too long to open, if the problem persist check app". any idea what this could be?

sthalik, 2017-02-27:
when does it happen?

hollywood156, 2017-02-27:
when i start the game. it only shows the message on the rift.

hollywood156, 2017-02-27:
i open the open track software first. then i open steam in vr mode. hit start on open track then i hit the play button for arma. that's how i was doing it on the first fix with no issues.

sthalik, 2017-03-01:
Which SDK version is it?

hollywood156, 2017-03-01:
1.4.0 I see there is a recent update. i will give it a try.

sthalik, 2017-04-06:
What message is it? The health warning?

sthalik, 2017-06-18:
No reply too long. :(

closed 2017-06-18 by sthalik