opentrack/opentrack #561: Unwanted pitch

blueeagle69, 2017-03-01:
Hi.
I am using Opentrack 2.3.0 with a PS3 Eye and DelanClip.
Every time I move yaw right, I getting unwanted pitch.
When turning my head right, I get pitch in one direction, then when turning back to centre and turning my head right again, I get pitch in another direction.
If I turn my head left, no pitch happens.
The yaw right is working fine of course, I just keep getting pitch thrown in.
I tried the latest version too with the same result.
I have also tried Dynamic Pose Resolution enabled & disabled.

Apart from this the head tracking works nice and responsive, and easily rivals the stupidly expensive commercial version.
It clearly something that is not quite right in my setup?
Never noticed the issue in FTNoIR.
Is my head on an angle or something :)

Len62, 2017-03-01:
I'll assume you have calibrated your set-up. You can try and add some deadspace in the pitch mapping. I needed to do this to prevent the same issue. OT is very sensitive to movement, as you have discovered.

blueeagle69, 2017-03-01:
Hi, yes I have calibrated a few times. The red crosshair is over the centre point.
I think that's correct.
I was thinking of adding a small dead zone to the pitch, but not a huge fan of dead zones in head tracking.
But it looks like I may have to.

Cheers.

Len62, 2017-03-01:
Are you using the Accela filter? You can play with the rotation filtering options there, it can be of some help.

blueeagle69, 2017-03-01:
Hi.
Yes I am.
I will have a play around with it and see later before Elite Dangerous calls me 😀

Len62, 2017-03-01:
See my edit to previous comment.

blueeagle69, 2017-03-01:
Thank you, in regards to the red point.
I will calibrate again and ensure the centre point is not obstructing the three points.
Oh well, that's one thing I had wrong 😉

blueeagle69, 2017-03-02:
Thanks Len62.

I looked at my calibration again. I looked at a video on YouTube.
I then re-calibrated again , but must have some how did the yaw left/right bit wrong because the red centre point appeared to the right of my head. So I simply added a negative value of the X axis to move it to the correct spot. Tried it in Elite Dangerous and this time I cannot spot any degree of pitch whilst yawing.
Thanks for all your help.
I knew there was something my 50yo brain had missed. LOL

blueeagle69, 2017-03-02:
The clip is on the right hand side of my head when facing the camera. So I take it the centre point should be where the head is 😊

Len62, 2017-03-02:
The OT display is not a mirror image. My clip is on the left side of my head, so the red cross appears to the left of the green crosses. Glad you got it sorted out.
[image]

blueeagle69, 2017-03-02:
Cheers Len.
My mistake.
I thought the red point was where the head was. 
I will correct it later.

Thanks again.

sthalik, 2017-03-04:
Thanks @Len62 :+1: 

@blueeagle69 the red cross (aka the pivot) is in the horizontal center of your head, like your neck.

blueeagle69, 2017-03-04:
Thanks.
The clip is on the right side of my head, and after calibration, the pivot point is on the right.
No more unwanted pitch though.

Len62, 2017-03-04:
That is where it should be.

blueeagle69, 2017-03-04:
Yeah, thanks again @Len62. 
I gathered that now.
Couldn't believe I have been using Opentrack this long with what I thought was a correctly calibrated   point tracker.
The only niggle I had was the pitch during yaw,  and you mentioning calibration fixed it.
Cheers.

closed 2017-12-02 by sthalik