opentrack/opentrack #562: Add "help" button.

okaryna, 2017-03-02:
It would be good to add some "help" button in the main window (eg. upper right corner) which leads to eg.
https://github.com/opentrack/opentrack/wiki/Quick-Start-Guide-(WIP)
Even better would be making of some local help (eventually with links to YouTube).
I know that for developers everything is obvious but for standard user or (especially) beginner such info like in quick start is valuable.

sthalik, 2017-03-04:
I completely agree. Ideally we'd link to the quickstart guide and a small dictionary of terms.

But I don't even have enough time to implement needed functionality, much less document it. You can see how descriptions in the options dialog are inaccurate, poorly worded, missing needed detail. The Russian translation has way better descriptions given a person dedicated time specifically to explain the functionality.

Try this:

- Install Qt. Go to http://download.qt.io/official_releases/qt/
- Clone the source code using Git or download and extract an archive from the project's main site
- Go to the "gui" subdirectory
- Open options-dialog.ui in the "Qt Designer" that you just installed
- Make a new tab by right-clicking and choosing the option, call it "Help"
- Put some text and links in there. You have rich text formatting.
- Send it back once you're satisfied. Either a pull request or just the file

@MathijsG you can use the procedure if you wish to pursue this instead.

Working on features and bugs burns me out a lot. The project forked in 2012-2013 and it's 2017. I'm the ONLY person working on core functionality for this time.

Sorry for the rant. You see how it is.

okaryna, 2017-03-04:
Believe me, I really appreciate your work. Without Opentrack, I would have to use Freetrack. And this is option I strongly want to avoid.
I will try to do as you described.

MathijsG, 2017-03-04:
What my concern, (besides having a quickstart which always is a good thing), is with this advanced help stuff, is that I don't have any numbers of people that are actually stuck all the time. Considering people must have already downloaded some app on their phone which needs to communicate to the computer, or they have already made some tracking clip which requires a certain skill. I do want to contribute my part on this project, I really like the software. But I don't really see many "heelp!! I don't know anything about this software" topics in here either. So is there a real need for better help/explanations? Don't get me wrong, to improve it is never wrong, but all our time is limited.

sthalik, 2017-03-04:
@MathijsG from my experience it's an advanced user that explains the software to new users. I saw on reddit and few forums people getting lost faced with the main screen. But there could be a "silent majority" not having issues with it. We'd need a poll or a few. Now it's not completely useless in the default configuration however, given that I no longer refuse to provide any sane defaults, heh.

There's a class of users that poke at things to find out how the software works, use precise google terms, have patience. But there are other users and it's not bad to be less technically inclined. I don't know the proportions.

I'll try running a poll.

sthalik, 2017-03-04:
Let's try the reddit part of Elite Dangerous community first.

https://www.reddit.com/r/EliteDangerous/comments/5xikxo/opentrack_poll_for_documentation_priority/

I'd wish for more exposure but what the hell.

sthalik, 2017-03-09:
Getting moderated and downvoted to hell rocks. If some of you can provide additional information or anecdotal evidence on the difficulty level I'd be glad.

MathijsG, 2017-03-09:
I think one might initially be a little blown away by all the technicalities and initial setups. But I think that the people who want it working out of the box with no hassle would already have bought TrackIR. If people really want to achieve low cost headtracking I guess they already searched through the tons of YouTube video's explaining the setup of Opentrack.

I am more fan of a forum-style interactive help when people get specifically stuck somewhere with it (did this before sometime on a Dutch forum), rather than writing big manuals where you don't really know if people ever read it, or even needed it.

Besides that we should minimize the amount of jargon if there are synonyms that people dó know.

And I'm still for this explanatory tooltip functionality Windows has:

[image]

Where terms like FPS would be explained on a way like this:
[image]

Where it would state something like "The amount of frames per second the camera will output. The more the more smooth the tracking process, but with a possible higher CPU load.?"

So in the end people would know the concept of a function, where a guide often just puts people to the neccessary option, but they don't really get to understand concepts. And they would all be very confused again when a button has been moved or renamed. This is what I see giving elderly people computer lessons.

sthalik, 2017-03-20:
I'm loaded with writing code whenever I have time for opentrack. I'm also biased toward already knowing what's what, and toward lazy half-assed descriptions.

@MathijsG you've been here for a long while. Please consider editing the .ui file as a pull request. There are also multiple half-assed descriptions in the English text itself. I understand it's a lot of work but it's not a programmer's job. It's like a person at work assumming I'd do both programming *and* QA.

sthalik, 2017-03-20:
Furthermore, you can do things partially. Adding some help will motivate further quality improvements.

MathijsG, 2017-03-20:
Okay, I will try to play around with the UI editor soon and check the UX.

okaryna, 2017-03-21:
I read this discussion and I want to explain one thing. Of course complex help would be awesome but my proposition was to add only one button who will launch github wiki "getting started" in a browser.
And I think from the point of a user it is enough.

sthalik, 2017-03-21:
i don't like buttons like that since they're prone to get pressed accidentally. it's not optimal either. we should get a welcome screen on the first run however.

MathijsG, 2017-03-30:
Do you have places where people would have been stuck? I sometimes visit the old free track forums to redirect people to opentrack instead of freetrack, because the latter is broken for modern systems and the development long time gone. I would happily visit some forums regularly to give specific help.

sthalik, 2017-03-30:
Some functionality's hard to understand in general. Look at "relative translation". Some get it by using, some not. Adding insult to injury, it was enabled by default accidentally.

Also, some of the keystrokes; game detection; mapping usage; Accela filter knobs.

sthalik, 2017-03-30:
How do you feel about removing non-auto-threshold mode from PT? If anything, the checkbox's confusing for the users.

okaryna, 2017-03-30:
Problems I have/had:
- differences between filters,
- sometimes need to check what is yaw/pitch/roll (yeah I know its easy, hovewer I have gaps between usage and sometimes need to check),
- which axis is real and which virtual,
- where to use different inputs (I mean what if PointTracker does not work).

MathijsG, 2017-03-30:
With places I meant actual places as in websites where people would be stuck :-). Concepts like relative translation things are inherently somewhat hard to understand. Maybe instead of using an actual technical term you can sometimes divert to naming such things like "While looking behind you in-game, you can move forward in-game by" and give the option "backward" or "forward".

closed 2018-08-03 by sthalik