opentrack/opentrack #568: mapping setting not apply when startup

flycrshish, 2017-03-07:
opentrack 2.3.1 , aruco + VJOY
mapping setting not apply when startup
1. adjust setting in mapping menu .
2. start tracking  , mapping setting is work.
3. close  opentrack , restart opentrack , start tracking. you can see the last mapping setting is not work.
4. open mapping menu , don't do any thing ,just press "OK"  .  then,the mapping setting is work again.

thank you and sorry for my english.

flycrshish, 2017-03-09:
at my step3 
after reopen opentrack ,if you do not open mapping menu first and start track direct.
mapping setting will not be apply until you open mapping menu and press "OK" .
example : 
put aruco marker in a Fixed position, 
enabled  options-Camera-"Use carmera center as translation offet"
mapping setting -  X Max translation "30mm"
start tracking ,opentrack show Raw tracker data X = 8   :  Game data X=22
 
mapping setting -  X Max translation "10mm"
opentrack show Raw tracker data X = 8   :  Game data X=63

close opentrack , restart opentrack ,don't do anything and just press start tracking
opentrack show Raw tracker data X = 8   :  Game data X=22  <--it's a default 30mm setting 

open mapping menu,  don't do anything and just press "OK"
opentrack show Raw tracker data X = 8   :  Game data X=63  <--10mm setting is appy

do that on two computer and get same situation.   thank

sthalik, 2017-03-09:
I put a point at X=0 Y=180, close/reopen, and start tracking. It correctly shows 180 following your procedure.

Can you confirm it with a tracker that doesn't work, like the joystick tracker with all input disabled? It could be a bug in aruco or some of the options.

flycrshish, 2017-03-09:
sorry , i don't have joystick now but i will test when i get one.
oh ,one thing i telled is Incorrect.
Game data value will change when you open mapping menu, no need to press "OK" .

sthalik, 2017-03-09:
You don't need to. It's an example of a tracker that can never output anything. You can also use "hatire arduino" if you don't have one connected.

sthalik, 2017-03-09:
Are you using a value for "max rotation"?

flycrshish, 2017-03-09:
no i just test X value.
Raw tracker data X = 0  : Game data X=0
move aruco marker to right
Raw tracker data X = 8  : Game data X=20
move aruco marker to left
Raw tracker data X = -8  : Game data X=-20

i think my aruco working normal .

sthalik, 2017-03-09:
See if today's build works now -- http://opentrack.misaki.pl

flycrshish, 2017-03-09:
new build is work, thank you very much ^_^

sthalik, 2017-03-09:
Thank you for the report and the final confirmation :+1: :cat: :cat2:

closed 2017-03-09 by sthalik

sthalik, 2017-03-09:
@flycrshish I released `2.3.1.4` with the fix included.