opentrack/opentrack #571: Names of UDP Receiver and UDP Sender are wrong and need to be switched [main exe]

adeldex, 2017-03-12:
So on some old rig I have my PS3 eye set up with opetrack 2.3.0 to take the load of my main processor and at first it was confusing to me because I wanted output to be UDP sender but couldn't select it. Now I have input as PointTracker 1.1 as usual and output as UDP receiver with my main pc's LAN address.
Then on my main pc on opentrack 2.3.1.4 its also reversed and now I have as input UDP sender and as output I have freetrack 2.0 as usual.
And it all works*. The main point is that the receiver sends to the sender. Only the labels are wrong.

*the x y z translations are weird but that's another issue

sthalik, 2017-03-20:
What's with the translation?

babbott1000, 2017-04-13:
Yes, I noticed this too.  I was really confused that the sender was in the input slot.  The free pie receiver or whatever was also in the input slot, so that is right but there is a sender in the input slot.  Would be nice if this would be fixed.

sthalik, 2017-04-20:
It's a matter of which side you're looking from. But I can appreciate your perspective, and that it's the more common one.

sthalik, 2017-04-20:
Done!

fuxingyang, 2018-01-26:
[image]
Hello,I have a similar problem with opentrack output setting.Can you help me? Thank you!Best wishes to you!

sthalik, 2018-01-26:
Responding in #736.

closed 2018-01-26 by sthalik